dentra/esphome-components #3: RTCGQ02LM Timeout and Idle Time

xbmcnut, 2021-09-24:
Hi there,

Got it working now thanks but I had a question about the `timeout` and `idle_time`. What are these used for and are they configurable? I know this sensor doesn't send a motion cleared event so there needs to be a way to tell HA the sensor has reset just wondering how these two entities function.

dentra, 2021-09-24:
In opposite to main sensor which can report their state immediately the `timeout` fires only when no motion detected as configured timeout in Mi Home app.
The `idle_time` additionally fires 2 times when no motion detected for a long time. It will report 0s at start motion, 120 when mo motion is 3 min ago and 300 when mo motion is 5 min ago.

dentra, 2021-09-24:
so you can configure only `timeout` directly in Mi Home app. 

`idle_time` is preconfigured in device firmware and at this moment I don't know how reconfigure it.

xbmcnut, 2021-09-24:
Brilliant, thank you. I would have never figured that out and was wondering why my 3 sensors were behaving differently.

closed 2021-09-24 by xbmcnut