dentra/esphome-components #4: ZMAi-90 V3 solution?

whulshof, 2021-10-02:
Hi, I read that you have solution for V3 of ZMAi-90 energy meter/switch. I replaced WB3S MCU by ESP8266 12E.
I have connection to unit, but the switch does not work and is off. The device has V9821S mcu for the measuring part.
If you could give me a hint, it would be great!

whulshof, 2021-10-03:
@Dentra, I already found your conversation in the home-assistant community with subject "Help with ZMAI-90 and ESPHome or Tasmota".
There you already posted the solution. And it works. Brilliant! I will close the issue after adding the ESPHome config for V3 of ZMAi-90:

substitutions:
  name: "ZMAi-90"
  node_name: zmai90

esphome:
  platform: ESP8266
  board: esp12e
  name: test

# Enable Home Assistant API
api:
  password: abcd1234

# Make sure you can upload new firmware OTA

ota:
  password: "2b76b9fd0e83315a866c18dcea62677c"

wifi:
  ssid: "SSID"
  password: "PASSWORD"
  # manual_ip:
    #   static_ip: 10.222.15.33
    #   gateway: 10.222.14.138
    #   subnet: 255.255.254.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Fallback Hotspot"
    password: "Qb2BgmhQxev8"
  fast_connect: true

captive_portal:

web_server:
  port: 80
  auth:
    username: willem
    password: erdr1ed

# Enable logging
logger:
  baud_rate: 0

uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
  #parity: EVEN

external_components:
  - source: github://dentra/esphome@tuya-raw-fix-standalone
    components: ["tuya"]
  - source: github://dentra/esphome-components

# Example configuration entry
time:
  - platform: homeassistant
    id: homeassistant_time

# Register the Tuya MCU connection
tuya:
  on_datapoint_update:
    - sensor_datapoint: 6
      datapoint_type: raw
      then:
        - lambda: |-
            id(voltage).publish_state((x[0] << 8 | x[1]) * 0.1);
            id(current).publish_state((x[3] << 8 | x[4]) * 0.001);
            id(power).publish_state((x[6] << 8 | x[7]));

sensor:
  # total
  - platform: "tuya"
    id: total
    name: "Energy Total"
    sensor_datapoint: 1
    accuracy_decimals: 3
    filters:
      - multiply: 0.001
    unit_of_measurement: "kWh"
    icon: "mdi:sigma"
    device_class: "energy"
  - platform: template
    id: voltage
    name: "Voltage"
    unit_of_measurement: "V"
    icon: "mdi:sine-wave"
    device_class: "voltage"
  - platform: template
    id: current
    name: "Current"
    unit_of_measurement: "A"
    icon: "mdi:current-ac"
    device_class: "current"
  - platform: template
    id: power
    name: "Active Power"
    unit_of_measurement: "W"
    accuracy_decimals: 0
    icon: "mdi:flash"
    device_class: "power"
  - platform: "energy_monitoring"
    power: power
    voltage: voltage
    current: current
    apparent_power:
      name: "Apparent Power"
    reactive_power:
      name: "Reactive Power"
    power_factor:
      name: "Power Factor"

switch:
   - platform: "tuya"
     name: "Power"
     switch_datapoint: 16
     icon: "mdi:power"

closed 2021-10-03 by whulshof

dentra, 2021-10-11:
Please vote for https://github.com/esphome/esphome/pull/1812 to include this path to esphome.