dentra/esphome-components #5: Problem with ESPHome firmware on ZMAi-90 WB3S V3

ArtKZ, 2021-10-05:
Hi!
Can you help me with the ZMAi-90 counter firmware? I bought several pieces with a WB3S chip, replaced the chip with an ESP-12F and used your firmware for version 2, but it does not work on this counter modification.

dentra, 2021-10-05:
Hi! Could you post Tuya initialization logs with list of datapoints

ArtKZ, 2021-10-05:
[16:58:15][I][app:106]: ESPHome version 2021.8.2 compiled on Oct  5 2021, 15:35:03
[16:58:15][C][wifi:499]: WiFi:
[16:58:15][C][wifi:359]:   SSID: [redacted]
[16:58:15][C][wifi:360]:   IP Address: 192.168.75.194
[16:58:15][C][wifi:362]:   BSSID: [redacted]
[16:58:15][C][wifi:363]:   Hostname: 'electricity-meter-cfcdd5'
[16:58:15][C][wifi:367]:   Signal strength: -70 dB ▂▄▆█
[16:58:15][C][wifi:371]:   Channel: 2
[16:58:15][C][wifi:372]:   Subnet: 255.255.255.0
[16:58:15][C][wifi:373]:   Gateway: 192.168.75.1
[16:58:15][C][wifi:374]:   DNS1: 192.168.70.1
[16:58:15][C][wifi:375]:   DNS2: (IP unset)
[16:58:15][C][uart_esp8266:075]: UART Bus:
[16:58:15][C][uart_esp8266:077]:   TX Pin: GPIO1
[16:58:15][C][uart_esp8266:080]:   RX Pin: GPIO3
[16:58:15][C][uart_esp8266:081]:   RX Buffer Size: 256
[16:58:15][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[16:58:15][C][uart_esp8266:084]:   Data Bits: 8
[16:58:15][C][uart_esp8266:085]:   Parity: NONE
[16:58:15][C][uart_esp8266:086]:   Stop bits: 1
[16:58:15][C][uart_esp8266:088]:   Using hardware serial interface.
[16:58:15][C][logger:189]: Logger:
[16:58:15][C][logger:190]:   Level: DEBUG
[16:58:15][C][logger:191]:   Log Baud Rate: 0
[16:58:15][C][logger:192]:   Hardware UART: UART0
[16:58:15][C][homeassistant.time:010]: Home Assistant Time:
[16:58:15][C][homeassistant.time:011]:   Timezone: 'TZ-6'
[16:58:15][D][tuya:407]: Sending WiFi Status
[16:58:15][C][tuya.sensor:028]: Tuya Sensor 'Electricity Meter CFCDD5 Energy Total'
[16:58:15][C][tuya.sensor:028]:   Device Class: 'energy'
[16:58:15][C][tuya.sensor:028]:   State Class: ''
[16:58:15][C][tuya.sensor:028]:   Unit of Measurement: 'kWh'
[16:58:15][C][tuya.sensor:028]:   Accuracy Decimals: 3
[16:58:15][C][tuya.sensor:028]:   Icon: 'mdi:sigma'
[16:58:15][C][tuya.sensor:029]:   Sensor has datapoint ID 17
[16:58:15][C][tuya.sensor:028]: Tuya Sensor 'Electricity Meter CFCDD5 Active Power'
[16:58:15][C][tuya.sensor:028]:   Device Class: 'power'
[16:58:15][C][tuya.sensor:028]:   State Class: ''
[16:58:15][C][tuya.sensor:028]:   Unit of Measurement: 'W'
[16:58:15][C][tuya.sensor:028]:   Accuracy Decimals: 1
[16:58:15][C][tuya.sensor:028]:   Icon: 'mdi:flash'
[16:58:15][C][tuya.sensor:029]:   Sensor has datapoint ID 19
[16:58:15][C][tuya.sensor:028]: Tuya Sensor 'Electricity Meter CFCDD5 Current'
[16:58:15][C][tuya.sensor:028]:   Device Class: 'current'
[16:58:15][C][tuya.sensor:028]:   State Class: ''
[16:58:15][C][tuya.sensor:028]:   Unit of Measurement: 'A'
[16:58:15][C][tuya.sensor:028]:   Accuracy Decimals: 3
[16:58:15][C][tuya.sensor:028]:   Icon: 'mdi:current-ac'
[16:58:15][C][tuya.sensor:029]:   Sensor has datapoint ID 18
[16:58:15][C][tuya.sensor:028]: Tuya Sensor 'Electricity Meter CFCDD5 Voltage'
[16:58:15][C][tuya.sensor:028]:   Device Class: 'voltage'
[16:58:15][C][tuya.sensor:028]:   State Class: ''
[16:58:15][C][tuya.sensor:028]:   Unit of Measurement: 'V'
[16:58:15][C][tuya.sensor:028]:   Accuracy Decimals: 1
[16:58:15][C][tuya.sensor:028]:   Icon: 'mdi:sine-wave'
[16:58:15][C][tuya.sensor:029]:   Sensor has datapoint ID 20
[16:58:15][C][energy_monitoring:011]: Energy monitoring sensors
[16:58:15][C][energy_monitoring:012]:   Apparent power 'Electricity Meter CFCDD5 Apparent Power'
[16:58:15][C][energy_monitoring:012]:     State Class: ''
[16:58:15][C][energy_monitoring:012]:     Unit of Measurement: 'VA'
[16:58:15][C][energy_monitoring:012]:     Accuracy Decimals: 1
[16:58:15][C][energy_monitoring:012]:     Icon: 'mdi:flash'
[16:58:15][C][energy_monitoring:013]:   Reactive power 'Electricity Meter CFCDD5 Reactive Power'
[16:58:15][C][energy_monitoring:013]:     State Class: ''
[16:58:15][C][energy_monitoring:013]:     Unit of Measurement: 'VAR'
[16:58:15][C][energy_monitoring:013]:     Accuracy Decimals: 1
[16:58:15][C][energy_monitoring:013]:     Icon: 'mdi:flash'
[16:58:15][C][energy_monitoring:014]:   Power factor 'Electricity Meter CFCDD5 Power Factor'
[16:58:15][C][energy_monitoring:014]:     Device Class: 'power_factor'
[16:58:15][C][energy_monitoring:014]:     State Class: ''
[... 40 lines cut ...]
[16:58:15][C][wifi_info:010]: WifiInfo SSID 'Electricity Meter CFCDD5 Connected SSID'
[16:58:15][C][wifi_info:011]: WifiInfo BSSID 'Electricity Meter CFCDD5 Connected BSSID'
[16:58:15][C][startup:013]: Startup Sensor 'Electricity Meter CFCDD5 Uptime'
[16:58:15][C][startup:013]:   Device Class: 'timestamp'
[16:58:15][C][startup:013]:   State Class: ''
[16:58:15][C][startup:013]:   Unit of Measurement: ''
[16:58:15][C][startup:013]:   Accuracy Decimals: 0
[16:58:15][C][startup:013]:   Icon: 'mdi:clock-start'
[16:58:15][C][tuya:027]: Tuya:
[16:58:15][C][tuya:040]:   Datapoint 1: int value (value: 8)
[16:58:15][C][tuya:036]:   Datapoint 6: raw (value: 09.A3.00.00.00.00.00.00 (8))
[16:58:15][C][tuya:046]:   Datapoint 10: bitmask (value: 0)
[16:58:15][C][tuya:038]:   Datapoint 12: switch (value: OFF)
[16:58:15][C][tuya:040]:   Datapoint 13: int value (value: 0)
[16:58:15][C][tuya:038]:   Datapoint 16: switch (value: ON)
[16:58:15][C][tuya:036]:   Datapoint 17: raw (value: 01.00.00.3C.03.00.01.08.04.00.00.96 (12))
[16:58:15][C][tuya:042]:   Datapoint 18: string value (value: 111111111111)
[16:58:15][C][tuya:054]:   Product: '{"p":"nzexeqam9qulajbf","v":"1.0.0","m":2}'
[16:58:18][D][tuya:268]: Datapoint 1 update to 8
[16:58:22][D][tuya:252]: Datapoint 6 update to 09.9F.00.00.00.00.00.00 (8)
[16:58:32][D][tuya:268]: Datapoint 1 update to 8
[16:58:34][D][sensor:131]: 'Electricity Meter CFCDD5 Uptime': Sending state 1633431296.00000  with 0 decimals of accuracy
[16:58:37][D][tuya:252]: Datapoint 6 update to 09.9A.00.00.00.00.00.00 (8)
[16:58:47][D][tuya:268]: Datapoint 1 update to 8
[16:58:52][D][tuya:252]: Datapoint 6 update to 09.96.00.00.00.00.00.00 (8)

ArtKZ, 2021-10-05:
I found out datapoint 16 is a relay

ArtKZ, 2021-10-05:
I found all the answers [here](https://community.home-assistant.io/t/help-with-zmai-90-and-esphome-or-tasmota/308554) and now the energy meter works great!
I am sorry for disturbing...

closed 2021-10-05 by ArtKZ