dentra/esphome-components #12: Device uptime sensor

to4ko, 2021-11-02:
is there is something wrong with uptime sensor?

[image]

the code 
`external_components:
  - source: github://dentra/esphome-components

sensor:
  - platform: startup
    name: Device Uptime`

dentra, 2021-11-02:
Do you sync the time with HA or a time server?

closed 2021-11-02 by to4ko

dentra, 2021-11-02:
You could sync your HA with time server, so will get the right results too

to4ko, 2021-11-02:
ш

> You could sync your HA with time server, so will get the right results too

it's already synced but the result is not what i was expected. once changed in esphome from ha time to time server - everything gets back to normal