dentra/esphome-components #16: validation error, platform: miot_thermogigro not found

renoye, 2021-12-13:
Trying to use your external component I got red cross says platform miot_thermogigro can not be found

`external_components:
  - source: github://dentra/esphome-components
    
sensor:
  - platform: miot_thermogigro
    # String (Required), device MAC-address.
    mac_address: "device-mac-address"
    # String, (Optional), device bind key. Will use "xiaomi_account" from "miot" if absent to automaticaly get the bindkey.
    bindkey: "device-bin-key"
    # String, (Optional), the name of sensor.
    name: "$name Temperature"
    # Sensor (Optional), Humidity, %
    humidity:
      name: "$name Humidity"
    # Sensor (Optional), Battey Level, %
    battery_level:
      name: "$name Battery Level"
    # Sensor (Optional), Battey Voltage, V
    battery_voltage:
      name: "$name Battery Voltage"`

If I do validation, I got this
`INFO Reading configuration /config/esphome/esphome-5th-device.yaml...
ERROR Unable to import component miot_thermogigro.sensor:
Traceback (most recent call last):
  File "/opt/esphome/esphome/loader.py", line 163, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/2f2a5be3/components/miot_thermogigro/sensor.py", line 3, in <module>
    from esphome.components import sensor, miot
  File "/config/esphome/.esphome/external_components/2f2a5be3/components/miot/__init__.py", line 8, in <module>
    from esphome.const import (
ImportError: cannot import name 'PLATFORM_ESP32' from 'esphome.const' (/opt/esphome/esphome/const.py)
Failed config

sensor.miot_thermogigro: [source /config/esphome/esphome-5th-device.yaml:37]
  
  Platform not found: 'sensor.miot_thermogigro'.
  platform: miot_thermogigro
  mac_address: device-mac-address
  bindkey: device-bin-key
  name: $name Temperature
  humidity: 
    name: $name Humidity
  battery_level: 
    name: $name Battery Level
  battery_voltage: 
    name: $name Battery Voltage`
Can you tell me what I did wrong? I assume this config is for esphome device yaml.

renoye, 2021-12-13:
forgot to mention I am using home assistant addon.

ESPHome
Current version: 2021.9.3 (Changelog)

dentra, 2021-12-14:
it only works with ESPHome 2021.11.0 and up

renoye, 2021-12-16:
I upgraded my esphome to 202112.0

and the device is working perfectly.

Thanks so much for your wonderful work.

closed 2021-12-16 by renoye