dentra/esphome-components #17: Is there a way to get bind key of RTCGQ02LM without Xiaomi hub?

scout1256, 2021-12-27:
Xiaomi Mi Smart Home Occupancy Sensor 2 (RTCGQ02LM)
Xiaomi Mi Smart Home Door/Window Sensor 2 (MCCGQ02HL)

MCCGQ02HL can pair directly through phone without hub.
Is there a go around?

dentra, 2022-01-04:
You don’t need a hub, just connect your device to Mi Home application. Then you can use `miot` integration with Xiaomi credentials to get bindkey automatically

dentra, 2022-05-01:
The problem is most likely solved. Issue will be closed soon.

timelyfiasco, 2022-06-23:
Hi I have a mccgq02hl I'm trying to install. @dentra can you help me get the bindkey of the device?

my yaml file is
```
...
external_components:
  - source: github://dentra/esphome-components
miot:
  xiaomi_account:
    username: #!secret xiaomi_username
    password: #!secret xiaomi_password
binary_sensor:
  - platform: miot_mccgq02hl
    # String (Required), device MAC-address.
    name: "$project Door/Window Sensor"
```

compile logs
```
esphome run brian/d_stealth_cam322testds.yaml
INFO Reading configuration brian/d_stealth_cam322testds.yaml...
INFO Generating C++ source...
INFO Got bindkey for E4:AA:EC:44:55:22 Camera ESP 011 Door/Window Sensor
INFO Compiling app...
...
```

Debug logs
```
[22:59:30][W][miot_decrypt:054]: E4AAEC4D5522 [098B] mbedtls_ccm_auth_decrypt failed: -15
[22:59:30][W][miot_decrypt:056]:    mac: e14:aa:ec:4d:55:212
[22:59:30][W][miot_decrypt:057]:    key: 53.93.4F.A4.91.AD.BC.DB.E8.70.0B.3C.D1.61.2A.B1 (16)
[22:59:30][W][miot_decrypt:058]:     iv: 22.55.44.EC.AA.E4.7B.09.59.00.00.00 (12)
[22:59:30][W][miot_decrypt:059]:    add: 11
[22:59:32][W][miot_decrypt:054]: E4AAEC445522 [098B] mbedtls_ccm_auth_decrypt failed: -15
INFO 192.168.1.62: Error while reading incoming messages: Error while reading data: [Errno 104] Connection reset by peer
INFO Disconnected from ESPHome API for 192.168.1.62
WARNING Disconnected from API
INFO 192.168.1.62: Ping Failed: Error while reading data: [Errno 104] Connection reset by peer
INFO Successfully connected to 192.168.1.62
[22:59:48][D][miot:146]: Got MiBeacon: 30.58.7B.09.5C.22.55.44.EC.AA.E4.08 (12)
[22:59:48][D][miot:150]:    [098B] E4:AA:EC:44:55:22 RSSI=-61 ▂▄▆█
[23:01:44][D][esp32_ble_tracker:723]: Found device F9:1C:0E:2C:88:FD RSSI=-90
[23:01:44][D][esp32_ble_tracker:744]:   Address Type: RANDOM
[23:02:30][D][miot:146]: Got MiBeacon: 30.58.8B.09.5D.22.55.4D.EC.AA.E4.08 (12)
[23:02:30][D][miot:150]:    [098B] E4:AA:EC:44:55:22 RSSI=-47 ▂▄▆█
```

dentra, 2022-06-23:
@timelyfiasco you already has got it but something went wrong when decoding a message. could you please provide a log with "mbedtls_ccm_auth_decrypt failed" but with "Got MiBeacon" before it?

dentra, 2022-06-23:
@timelyfiasco I also see something strange in your configuration. It is two configured MCCGQ02HL, with MACs E4:AA:EC:**4D**:55:22 and E4:AA:EC:**44**:55:22, but only second is configured via miot.xiaomi_account. The log has the product ID 097B instead of 098B and this is possibly the reason. Do you have a box with your device model number?

timelyfiasco, 2022-06-24:
Hi @dentra here's the whole portion. I thought I should fudge the mac address that's why you found a mistake. There's only one device running. Here's the real log:
```
[23:14:27][D][miot:146]: Got MiBeacon: 58.58.8B.09.70.22.55.4D.EC.AA.E4.44.55.01.32.00.00.00.CE.FD.03.FD (22)
[23:14:27][D][miot:150]:    [098B] (encrypted) E4:AA:EC:4D:55:22 RSSI=-34 ▂▄▆█
[23:14:27][W][miot_decrypt:054]: E4AAEC4D5522 [098B] mbedtls_ccm_auth_decrypt failed: -15
[23:14:27][W][miot_decrypt:056]:    mac: e4:aa:ec:4d:55:22
[23:14:27][W][miot_decrypt:057]:    key: 53.93.4F.A4.91.AD.BC.DB.E8.70.0B.3C.D1.61.20.A1 (16)
[23:14:27][W][miot_decrypt:058]:     iv: 22.55.4D.EC.AA.E4.8B.09.70.00.00.00 (12)
[23:14:27][W][miot_decrypt:059]:    add: 11
[23:14:27][W][miot_decrypt:060]:    tag: CE.FD.03.FD
[23:14:27][W][miot_decrypt:061]:   data: 44.55.01.32
```

dentra, 2022-06-24:
@timelyfiasco There really is some kind of problem with the bindkey. Is your device currently connected to the Mihome app and is it working correctly? This is a necessary condition for the correct operation of automatic bindkey collection. In your sample config username and password secrets are commented, I hope all things are correct in your real config.

You could try to change bindkey by repairing your device to MiHome application and then get the key again, but you also need to temporary disable cache by setting update_interval to 0s for xiaomi_account:
```yaml
miot:
  xiaomi_account:
    username: !secret xiaomi_username
    password: !secret xiaomi_password
    update_interval: 0s
```

timelyfiasco, 2022-06-24:
Yes I can see the status of the device from the "xiaomi home" app when I play with the sensor. But I cant see the device when I log in to https://account.xiaomi.com thru firefox. It is only visible in the android app. 
The credentials are not commented out in my config, I hardcoded it for now. Is the region important? I commented out the region in the config.

timelyfiasco, 2022-06-24:
@dentra Would it be easier if I try to extract the bindkey? I dont have experience in network sniffing though. But I did find a guide for it. Do you suggest that I follow this guide? https://blog-vyoralek-cz.translate.goog/iot/esp32-a-esphome-jako-xiaomi-bluetooth-gateway-pro-ha/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en-US&_x_tr_pto=wapp

dentra, 2022-06-27:
`xiaomi_account` uses https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor to extract bindkeys. You can find a cache in .esphome/build/<node_name>/.xiaomi-devices.json. Or you can use Xiaomi-cloud-tokens-extractor itself. Also you can read https://custom-components.github.io/ble_monitor/faq#my-sensors-ble-advertisements-are-encrypted-how-can-i-get-the-key and find another way

timelyfiasco, 2022-06-27:
Hi @dentra my .../.esphome/build/espcam_01test1/.xiaomi-devices.json is similar to the result obtained from running `python3 token_extractor.py` but im getting `mbedtls_ccm_auth_decrypt failed: -15`

When I log in to mi home app, it asks me to upgrade the firmware of the sensor. Should I upgrade the firmware? 

Logs:
```

Logging in...
Logged in.

Devices found for server "cn":
   ---------
   NAME:     Door and window sensor
   ID:       blt.1.1anln0v6l4g00
   BLE KEY:  53934fa491adbcdbe8700b3cd16120a1
   MAC:      E4:AA:EC:4D:55:22
   TOKEN:    b37e3c2dbac921385244f569
   MODEL:    isa.magnet.dw2hl
   ---------

No devices found for server "de".
No devices found for server "us".
No devices found for server "ru".
Devices found for server "tw":
   ---------
   NAME:     Door sensor 1st
   ID:       blt.1.1anln0v6l4g00
   BLE KEY:  3f517aa8db0cc2f2b39761df84015ed3
   MAC:      E4:AA:EC:4D:55:22
   TOKEN:    f0c8382e6b2091c4853357e9
   MODEL:    isa.magnet.dw2hl
   ---------

Devices found for server "sg":
   ---------
   NAME:     Door sensor 1st
   ID:       blt.1.1anln0v6l4g00
   BLE KEY:  3f517aa8db0cc2f2b39761df84015ed3
   MAC:      E4:AA:EC:4D:55:22
   TOKEN:    f0c8382e6b2091c4853357e9
   MODEL:    isa.magnet.dw2hl
   ---------

Devices found for server "in":
   ---------
   NAME:     Door sensor 1st
   ID:       blt.1.1anln0v6l4g00
   BLE KEY:  3f517aa8db0cc2f2b39761df84015ed3
   MAC:      E4:AA:EC:4D:55:22
   TOKEN:    f0c8382e6b2091c4853357e9
   MODEL:    isa.magnet.dw2hl
   ---------

No devices found for server "i2".

Press ENTER to finish

```
```
xiaomi-devices.json
[
  {
    "did": "blt.1.1anln0v6l4g00",
    "mac": "E4:AA:EC:4D:55:22",
    "name": "Door and window sensor",
    "model": "isa.magnet.dw2hl",
    "token": "b37e3c2dbac921385244f569",
    "beaconkey": "53934fa491adbcdbe8700b3cd16120a1"
  },
  {
    "did": "blt.1.1anln0v6l4g00",
    "mac": "E4:AA:EC:4D:55:22",
    "name": "Door sensor 1st",
    "model": "isa.magnet.dw2hl",
    "token": "f0c8382e6b2091c4853357e9",
    "beaconkey": "3f517aa8db0cc2f2b39761df84015ed3"
  },
  {
    "did": "blt.1.1anln0v6l4g00",
    "mac": "E4:AA:EC:4D:55:22",
    "name": "Door sensor 1st",
    "model": "isa.magnet.dw2hl",
    "token": "f0c8382e6b2091c4853357e9",
    "beaconkey": "3f517aa8db0cc2f2b39761df84015ed3"
  },
  {
    "did": "blt.1.1anln0v6l4g00",
    "mac": "E4:AA:EC:4D:55:22",
    "name": "Door sensor 1st",
    "model": "isa.magnet.dw2hl",
    "token": "f0c8382e6b2091c4853357e9",
    "beaconkey": "3f517aa8db0cc2f2b39761df84015ed3"
  }
]
```

dentra, 2022-06-28:
Something is very strange with your configuration, for some reason your device is tied to 4 servers but not one. also, the bindkey is now different from what was in the log above, did you re-bind? So I can't try it myself. In any case, try to manually put one of these bindkeys:
53934fa491adbcdbe8700b3cd16120a1 (from CN server)
3f517aa8db0cc2f2b39761df84015ed3 (from other 3 servers)

timelyfiasco, 2022-07-08:
Hi @dentra, I got the time to test it again today. I got it working when I used the 2nd bindkey. The bindkey from CN didnt work for me. I get 3 servers of the same bindkey when I added a device to my region of choice. Thanks for your help!

Another question: Do battery level and battery voltage work? It shows N/A on my device.

dentra, 2022-07-08:
@timelyfiasco It may take some time to show the battery status. Battery voltage is calculated value and depends on battery level.

closed 2022-09-14 by stale[bot]