dentra/esphome-components #18: ZMAI-1: ¿Status?

jasimancas, 2022-01-18:
Hi, I have a couple of questions:

Is it possible to know the status of the switch?
Is it possible to act on it from HA or from webserver?

The supply has been cut and when the switch returns to the zmai it stays off, having to manually press the button to return the supply, so I wonder if you can see that information.

dentra, 2022-01-20:
As far I know it is a know problem of this device. If you turn off the switch, it will completely turn off the power, including device itself and no way to turn it on because it is offline. If I am not right and you could suggest a solution I'm will try to implement it.

Personally, I only use v2 as an energy meter in and don't even publish the switch to HA.

jasimancas, 2022-01-20:
The problem is that if the "switch" was published in ESP, when the power is cut and returns to the second I could from HA switch the switch to ON and let the power pass to the rest of the house, as both HA and the AP are protected with a UPS that holds on about 12 minutes, plenty of time if it is a temporary cut.

dentra, 2022-01-21:
Are you trying to say that the switch is always stay off after the main power is on back?

jasimancas, 2022-01-23:
Yes, just this.

dentra, 2022-01-24:
I added the restore_mode parameter to the main switch.

jasimancas, 2022-01-24:
It Work!!

Is possible to view status in webservice?

dentra, 2022-01-28:
sorry, i do not understand again :( status of the switch - it is switch itself. If you need additional sensor based on switch status you could create binary_sensor template and link it with the main switch.

closed 2022-01-31 by jasimancas