dentra/esphome-components #19: Many compile errors

JeeCee1, 2022-01-27:
Hi, thank you for this component!

I am running ESPhome 2022.1.2 on Supervised HAS (Debian) 2021.12.10.

My yaml file:
```
esp32_ble_tracker:

external_components:
  - source: github://dentra/esphome-components
  
binary_sensor:
  - platform: miot_mccgq02hl
    mac_address: ${MCCGQ02HL_1_mac}
    bindkey: ${MCCGQ02HL_1_bindkey}
    name: ${MCCGQ02HL_1_name}
    light:
      name: ${MCCGQ02HL_1_name} Light
    alert:
      name: ${MCCGQ02HL_1_name} Alert
    battery_level:
      name: ${MCCGQ02HL_1_name} Bat
    battery_voltage:
      name: ${MCCGQ02HL_1_name} Bat Volt.

sensor:
- platform: ble_rssi
  mac_address: ${MCCGQ02HL_1_mac}
  name: ${MCCGQ02HL_1_name} BLE RSSI
  entity_category: "diagnostic"
```
and in the parent file:
```
substitutions:
  MCCGQ02HL_1_mac: "E4:AA:EC:45:10:B4"
  MCCGQ02HL_1_bindkey: "edded93604df960caed5d47f"
  MCCGQ02HL_1_name: "MCCGQ02HL Door 1 "
```
The errors are many (20-30), I'll list the first few
`src/esphome/components/miot/miot_decrypt.h:13:31: error: 'MiotListener' does not name a type
bool decrypt_mibeacon45(const MiotListener *listener, MiBeacon &mib);`

`src/esphome/components/miot/miot_decrypt.h:13:55: error: 'MiBeacon' has not been declared
 bool decrypt_mibeacon45(const MiotListener *listener, MiBeacon &mib);`

`src/esphome/components/miot/miot_dump.cpp:8:40: error: 'MiBeacon' does not name a type
 void dump(const char *const TAG, const MiBeacon &mib) {`

`src/esphome/components/miot/miot_dump.cpp:9:43: error: request for member 'product_id' in 'mib', which is of non-class type 'const int' 
ESP_LOGD(TAG, "  Product ID: %04X", mib.product_id);`

`src/esphome/components/miot/miot_dump.cpp:11:57: error: request for member 'frame_control' in 'mib', which is of non-class type 'const int'
ESP_LOGD(TAG, "    is_encrypted      : %s", YESNO(mib.frame_control.is_encrypted));`

`src/esphome/components/miot/miot_dump.cpp:22:16: error: 'mac_reverse' was not declared in this scope
     auto mac = mac_reverse(mib.mac_address);`

`src/esphome/components/miot/miot_dump.cpp:23:37: error: expected ')' before 'ESP_BD_ADDR_STR'
     ESP_LOGD(TAG, "  MAC address: " ESP_BD_ADDR_STR, mac[5], mac[4], mac[3], mac[2], mac[1], mac[0]);`

I have cleaned all build files, but the errors persist.

Most likely I am doing something wrong, can you please help?

dentra, 2022-01-28:
looks like corrupted repository cache (1) or you building your project using esp-idf (2). 

1. try add `refresh: 0` option to external component, to get a fresh copy. (it will slow down every build, so you need remove it later)
2. switch to arduino platform

here is a fresh new config with your data that I tested:
```yaml
substitutions:
  MCCGQ02HL_1_mac: "E4:AA:EC:45:10:B4"
  MCCGQ02HL_1_bindkey: "edded93604df960caed5d47f"
  MCCGQ02HL_1_name: "MCCGQ02HL Door 1"
  
esphome:
  name: test-mccgq02hl

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "96e5ccd580aa9db354bebb6bd89e6bcb"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test-Mccgq02Hl Fallback Hotspot"
    password: "vp1CC4goYGLz"

captive_portal:

esp32_ble_tracker:

external_components:
  - source: github://dentra/esphome-components
  
binary_sensor:
  - platform: miot_mccgq02hl
    mac_address: ${MCCGQ02HL_1_mac}
    bindkey: ${MCCGQ02HL_1_bindkey}
    name: ${MCCGQ02HL_1_name}
    light:
      name: ${MCCGQ02HL_1_name} Light
    alert:
      name: ${MCCGQ02HL_1_name} Alert
    battery_level:
      name: ${MCCGQ02HL_1_name} Bat
    battery_voltage:
      name: ${MCCGQ02HL_1_name} Bat Volt.

sensor:
- platform: ble_rssi
  mac_address: ${MCCGQ02HL_1_mac}
  name: ${MCCGQ02HL_1_name} BLE RSSI
  entity_category: "diagnostic"
```

dentra, 2022-01-28:
also I have made an update to remove all deprecation warnings

JeeCee1, 2022-01-30:
Hi, thanks for the quick reply!

It was indeed the esp-idf framework.
How did you come to this conclusion / how could I have debugged this?

I am using the esp-idf framework because it reduces the code site significantly. Is there any chance you can make it work with that framework, or give me some pointers where to look so I can try to port it?

PS1: I was unable to get any data from the devices, it turned out that for the `BINDKEY` I needed to use `BLE KEY` in stead of `TOKEN` (I am not using your cloud retrieval option). Maybe add his to the docs?
PS2: `battery_level` and `battery_voltage` don't return any value?

dentra, 2022-01-30:
I adapt it to IDF as soon as ESPHome will start support bluetooth stack on it.

The docs nowhere says using token, only bindkey. It is common for all Xiaomi BT devices. I made cloud retrieval option especially for easy configuration, it is safe and run only on your build PC and only at build time, after that it injects retrived values to firmware.

Battery should report their value once 10-15 minutes.

dentra, 2022-01-31:
hmm... I missed something) added the ability to compile under esp-idf, but did not test

JeeCee1, 2022-02-02:
Cool, esp-idf framework works, thanks a lot!

Indeed, I needed to be a bit more patient for the battery reporting.

closed 2022-02-02 by JeeCee1