dentra/esphome-components #21: Integrate Yeelight Remote Control 1S (YLAI003)

gianlucaf81, 2022-03-03:
I ask you for a clarification on how to integrate this button in HA. I have configured everything correctly, in ESPHome log I see the single, double and long click. How do I use the event in HA? Only the battery status appears (yes work!). Thank you.

https://user-images.githubusercontent.com/39622729/156578768-9a425233-2362-404e-9f7a-881996014661.mp4

dentra, 2022-03-03:
This component allow you make automations on device events. For example you can turn on some light when button click event occurs. Or if you want you can create template sensor and change it state in automation and this sensor will be published in HA.

gianlucaf81, 2022-03-03:
ok i think this is the solution. https://esphome.io/guides/automations.html#automation
I guess it might be a nuisance for you but if you could include a small example of typical use for HA it would help beginners tremendously.
anyway thanks, I study this thing a bit.

samson-fu, 2022-06-20:
> 

I am beginner to HA and ESPHome, I've read the ESP automations document as suggested by @gianlucaf81. But I am still unable to add the YeeLight Dimmer Control to Home Assistant as a device with all the triggers. Is there any example / demo or tutorial for better understanding about the intergration? Thanks in advance!

gianlucaf81, 2022-06-20:
This is the code for the button. But if I can give you some advice use the Xiaomi Gateway 3. it works much better.
[button.yaml.txt](https://github.com/dentra/esphome-components/files/8938681/button.yaml.txt)

dentra, 2022-06-21:
@gianlucaf81 hide passwords in your example

gianlucaf81, 2022-06-22:
> @gianlucaf81 hide passwords in your example

ESP has not been online for a long time but thank you!

closed 2022-09-06 by stale[bot]