dentra/esphome-components #22: Buliding error on ESPHOME 2022.3

itanczos, 2022-03-17:
```
ERROR Unable to load component miot_mccgq02hl.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/2f2a5be3/components/miot_mccgq02hl/binary_sensor.py", line 27, in <module>
    ): binary_sensor.device_class,
AttributeError: module 'esphome.components.binary_sensor' has no attribute 'device_class'
ERROR Unable to load component miot_mccgq02hl.binary_sensor:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/esphome/.esphome/external_components/2f2a5be3/components/miot_mccgq02hl/binary_sensor.py", line 27, in <module>
    ): binary_sensor.device_class,
AttributeError: module 'esphome.components.binary_sensor' has no attribute 'device_class'
```

dentra, 2022-03-17:
Thanks. I haven't adapted to the new version yet. In progress.

dentra, 2022-03-18:
Please check

itanczos, 2022-03-19:
Thank you! Working fine! :)

closed 2022-03-19 by itanczos