dentra/esphome-components #24: miot_sjws01lm ESP sees devices, home assistant only shows

kyfehr, 2022-04-24:


dentra, 2022-04-25:
Try press a button on device or wait some time. This device sends data very rarely

kyfehr, 2022-04-25:
Thank you for the reply. I have had this set up and running for about 2 days and still no data comes through. Also tried to simulate a flood and no data. Not expecting a fix, just want to share that it seems to not be working as intended. On ESPHome 2022.4.0 and Home Assistant 2022.4.6 in case that is useful.

dentra, 2022-04-25:
Tested with EspHome 2022.4.0 and HA 2022.4.7 and it works as expected. 

Are your device is connected to MiHome and is it changes state in it?

Download latest version of my components or use github link in external component section.

Please re-check MAC address of your device in yaml configuration.

Also please be sure that you are using at least 1A power supply (unfortunately 0.7 may not be enough for BLE) and try to get data with 1m distance (for this test). Please provide debug logs when you push a button and emulate a water leak.

kyfehr, 2022-04-28:
I ended up rebuilding a few times, and it is now working. I think it is related to the device, but they do not update very quickly.

mamartins, 2022-04-30:
Is there a way to get the bindkey without the hub? I don’t have one :(

mamartins, 2022-04-30:
I'm having the same issue!

dentra, 2022-05-01:
@mamartins please file a new issue with description of your problem. Possibly it not the same.