dentra/esphome-components #25: Can't compile the code with new esphome versions

inorhan, 2022-05-26:
Compile errors with esphome: 2022.5.1

src/esphome/components/energy_tariffs/energy_tariffs.cpp: In member function 'virtual void esphome::energy_tariffs::EnergyTariffs::setup()':
src/esphome/components/energy_tariffs/**energy_tariffs.cpp:37:27: error: 'class esphome::number::Number' has no member named 'set'**
   37 |       this->time_offset_->set(loaded);
      |                           ^~~
Compiling /data/xxxxx/.pioenvs/xxxxx/src/esphome/components/number/automation.cpp.o
*** [/data/xxxxx/.pioenvs/xxxxx/src/esphome/components/energy_tariffs/energy_tariffs.cpp.o] Error 1

dentra, 2022-05-26:
This is due breaking changes in ESPHome 2022.5.0. Will fix it.

dentra, 2022-05-26:
fixed

closed 2022-05-26 by dentra

inorhan, 2022-05-26:
It worked, Thanks!

Had to use "refresh: 0s" to force the update the first time.

external_components:
  - source: github://dentra/esphome-components
    #refresh: 0s