dentra/esphome-components #28: Changing the mode and scene on a  EraClean CW-BS01 doesn't seem to work

mihsu81, 2022-07-09:
Hi @dentra,

I'm trying to control my EraClean CW-BS01. Whichever commands I send (setting the mode, the type of scene), be them from HA or directly from the ESP32 webserver don't actually change any settings on the device.
Is this caused by the fact that it is a work in progress? I can see from the logs that the command sent is still with the old settings (as exemplified in the first 3 lines of the below log).

Thanks in advance.

```
[10:11:01][D][select:062]: 'FRIDGE-AIR-PURIFIER Mode' - Setting
[10:11:01][D][select:115]: 'FRIDGE-AIR-PURIFIER Mode' - Set selected option to: Anion
[10:11:01][D][select:015]: 'FRIDGE-AIR-PURIFIER Mode': Sending state Keep - Freshing (index 1)
[10:11:08][D][ble_client:047]: Found device at MAC address [EC:4D:3E:BE:A7:6D]
[10:11:08][D][iBeacon:101]: UUID: 50765CB7-D9EA-4E21-99A4-FA879613A492
[10:11:08][I][ble_client:085]: Attempting BLE connection to ec:4d:3e:be:a7:6d
[10:11:10][I][ble_client:170]: Service UUID: 0x1801
[10:11:10][I][ble_client:171]:   start_handle: 0x1  end_handle: 0x4
[10:11:10][I][ble_client:383]:  characteristic 0x2A05, handle 0x3, properties 0x20
[10:11:10][I][ble_client:170]: Service UUID: 0x1800
[10:11:10][I][ble_client:171]:   start_handle: 0x5  end_handle: 0xd
[10:11:10][I][ble_client:383]:  characteristic 0x2A00, handle 0x7, properties 0x2
[10:11:10][I][ble_client:383]:  characteristic 0x2A01, handle 0x9, properties 0x2
[10:11:10][I][ble_client:383]:  characteristic 0x2A04, handle 0xb, properties 0x2
[10:11:10][I][ble_client:383]:  characteristic 0x2AA6, handle 0xd, properties 0x2
[10:11:10][I][ble_client:170]: Service UUID: 0xFE95
[10:11:10][I][ble_client:171]:   start_handle: 0xe  end_handle: 0x1c
[10:11:10][I][ble_client:383]:  characteristic 0x0004, handle 0x10, properties 0x2
[10:11:10][I][ble_client:383]:  characteristic 0x0010, handle 0x12, properties 0x14
[10:11:10][I][ble_client:383]:  characteristic 0x0019, handle 0x15, properties 0x14
[10:11:10][I][ble_client:383]:  characteristic 0x0017, handle 0x18, properties 0x18
[10:11:10][I][ble_client:383]:  characteristic 0x0018, handle 0x1b, properties 0x14
[10:11:10][I][ble_client:170]: Service UUID: 00000100-0065-6C62-2E74-6F696D2E696D
[10:11:10][I][ble_client:171]:   start_handle: 0x1d  end_handle: 0xffff
[10:11:10][I][ble_client:383]:  characteristic 00000101-0065-6C62-2E74-6F696D2E696D, handle 0x1f, properties 0x4
[10:11:10][I][ble_client:383]:  characteristic 00000102-0065-6C62-2E74-6F696D2E696D, handle 0x21, properties 0x10
[10:11:10][D][miot_standard_auth_client:090]: state=0xB (11)
[10:11:10][D][miot_standard_auth_client:090]: state=0xB (11)
[10:11:10][D][miot_standard_auth_client:090]: state=0x0 (0)
[10:11:10][D][miot_standard_auth_client:090]: state=0xD (13)
[10:11:10][D][miot_standard_auth_client:090]: state=0x0 (0)
[10:11:10][D][miot_standard_auth_client:090]: state=0xC (12)
[10:11:10][D][miot_standard_auth_client:090]: state=0xC (12)
[10:11:10][D][miot_standard_auth_client:090]: state=0xA (10)
[10:11:10][D][miot_standard_auth_client:090]: state=0xA (10)
[10:11:10][I][miot_standard_auth_client:062]: Login Success
[10:11:10][D][miot_cwbs01.api:156]: Request state
[10:11:10][D][miot_cwbs01.api:206]: Set mode 1
[10:11:10][D][text_sensor:067]: 'FRIDGE-AIR-PURIFIER Version': Sending state '1.1.1_0008'
[10:11:10][D][miot_cwbs01.api:073]: Got state: 5B.F5.FF.3F
[10:11:10][D][select:015]: 'FRIDGE-AIR-PURIFIER Mode': Sending state Keep - Freshing (index 1)
[10:11:10][D][select:015]: 'FRIDGE-AIR-PURIFIER Scene': Sending state Middle refrigerator (150L - 300L) (index 1)
[10:11:10][D][sensor:125]: 'FRIDGE-AIR-PURIFIER Battery Level': Sending state 90.00000 % with 0 decimals of accuracy
[10:11:10][I][ble_client:075]: [ec:4d:3e:be:a7:6d] Disabling BLE client.
```

dentra, 2022-07-14:
Hi @mihsu81 !

I don't see any change requests in the log only received current state.

mihsu81, 2022-07-16:
Hi @dentra,
Sorry for the delay. I've attached the verbose logs below.
[logs_ble-gateway01_logs.txt](https://github.com/dentra/esphome-components/files/9126440/logs_ble-gateway01_logs.txt)

dentra, 2022-07-18:
thanks, I found and made small fixes, please update

dentra, 2022-07-19:
@mihsu81 as user reported me in telegram the component now works properly

dentra, 2022-07-19:
also fixed the README file

vladeha, 2022-07-19:
> @mihsu81 as user reported me in telegram the component now works properly

Yes, it works. Thank you for your support!

mihsu81, 2022-07-19:
@dentra It seems to work but there's another issue. From Home Assistant, when you change the mode or scene, it appears changed for a few seconds then reappears as the previous mode or scene. On the device it is actually changed. If again you change to another mode or scene, it displays the preciously selected mode or scene.
I'm not currently at home so I cannot check from the web server of the device.

dentra, 2022-07-19:
@mihsu81 I will look at

dentra, 2022-07-22:
@mihsu81 I made some changes. In my installation it works fine. Could you please check yours?

mihsu81, 2022-07-23:
Hi @dentra,
It looks like the issue is indeed solved.
I'll be able to confirm 100% tomorrow evening when I get back home and I'll have physical access to the device.
Thank you. 😊

mihsu81, 2022-07-24:
@dentra I can confirm it is working properly and no issues are encountered whatsoever.
Thanks again.

closed 2022-07-24 by mihsu81

mihsu81, 2022-07-25:
@dentra By the way, do you have plans to also implement the Timer switches?
Thanks.

dentra, 2022-07-25:
This is possible, but I do not know why it is, if all this logic can be implemented on the ESP or HA side. 
Anyway, you can feel free to suggest a control options with ESPHome to support this feature. Please in separate issue.

DivanX10, 2023-08-10:
Thanks for this component. EraClean Intelligent Deodorizer Max is perfectly controlled and the sensors work. I want to add for reference, so that there is an understanding if someone also receives the Failed login: 0x0023 error. I ran into a problem and here I did not find an answer. Then I realized the reason why the Failed login error occurred: 0x0023. EraClean Intelligent Deodorizer Max does not support multiple connections and if the phone where MiHome is installed and bluetooth is enabled, then EraClean Intelligent Deodorizer Max will connect to the phone by default and esp32 will lose contact with EraClean Intelligent Deodorizer Max. In the logs, ESPHome will receive the Failed login error: 0x0023

Log with connection error, where we see Failed login: 0x0023
```
[12:15:20][D][esp32_ble_client:048]: [0] [XX:XX:XX:XX:XX:XX] Found device
[12:15:20][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[12:15:20][I][esp32_ble_client:064]: [0] [XX:XX:XX:XX:XX:XX] 0x00 Attempting BLE connection
[12:15:23][I][esp32_ble_client:196]: [0] [XX:XX:XX:XX:XX:XX] Connected
[12:15:23][D][esp32_ble_tracker:245]: Starting scan...
[12:15:23][D][miot_standard_auth_client:090]: state=0xB (11)
[12:15:24][D][miot_standard_auth_client:090]: state=0xB (11)
[12:15:24][D][miot_standard_auth_client:090]: state=0x0 (0)
[12:15:24][D][miot_standard_auth_client:090]: state=0xD (13)
[12:15:24][D][miot_standard_auth_client:090]: state=0x0 (0)
[12:15:24][D][miot_standard_auth_client:090]: state=0xC (12)
[12:15:24][D][miot_standard_auth_client:090]: state=0xC (12)
[12:15:24][W][miot_standard_auth_client:212]: Fail compare
[12:15:24][D][miot_standard_auth_client:090]: state=0xA (10)
[12:15:24][D][miot_standard_auth_client:090]: state=0xA (10)
[12:15:24][W][miot_standard_auth_client:069]: Failed login: 0x0023
[12:20:07][I][ota:113]: Boot seems successful, resetting boot loop counter.

```

Whatever happens, there is Bluetooth support on the phone with the MiHome connected, and the esp 32 can easily establish a connection with the EraClean Max intelligent deodorant.

```
14:24:12	[D]	[esp32_ble_client:048]	
[0] [XX:XX:XX:XX:XX:XX] Found device
14:24:12	[D]	[esp32_ble_tracker:213]	
Pausing scan to make connection...
14:24:12	[I]	[esp32_ble_client:064]	
[0] [XX:XX:XX:XX:XX:XX] 0x00 Attempting BLE connection
14:24:14	[I]	[esp32_ble_client:196]	
[0] [XX:XX:XX:XX:XX:XX] Connected
14:24:14	[D]	[esp32_ble_tracker:245]	
Starting scan...
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xB (11)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xB (11)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0x0 (0)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0x0 (0)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xD (13)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0x0 (0)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xC (12)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xC (12)
14:24:15	[D]	[miot_standard_auth_client:090]	
state=0xA (10)
14:24:15	[D]	[miot_standard_auth_client:090]	
state=0xA (10)
14:24:15	[I]	[miot_standard_auth_client:062]	
Login Success
```

dentra, 2023-08-12:
@DivanX10, Thanks for your experience. Could you suggest changes to the documentation for this component by PR or just as an new issue?