dentra/esphome-components #29: Add support for yunmi.kettle.v12 kettle

tannisroot, 2022-07-22:
Hi! I tried your custom component with my Xiaomi Mi Smart Kettle Pro (sold in Europe), which looks a lot like the model specified as supported in the readme, but unfortunately it doesn't seem to work. I am not 100% sure i didn't just set it up wrong, but it doesn't look like I did.
Here's some info I figured could be useful.

My config section related to miot kettle platform: 

```esp32_ble_tracker:

external_components:
  - source: github://dentra/esphome-components

miot:
  xiaomi_account:
    username: !secret xiaomi_username
    password: !secret xiaomi_password
    servers: "ru"
    update_interval: 4h
  
binary_sensor:
  - platform: miot_kettle
    mac_address: "A4:C1:38:82:C9:35"
    name: "Kettle Water Boil Power"
    temperature:
      name: "Kettle Water Boil Temperature"
```
When I compile firmware, I see this in the beginning, so that part must be working.

```
INFO Got bindkey for A4:C1:38:82:C9:35 Kettle Water Boil Power
```

With verbose logging, I can see this about my kettle:

```
[14:07:45][VV][esp32_ble_tracker:500]: Parse Result:
[14:07:45][VV][esp32_ble_tracker:517]:   Address: A4:C1:38:82:C9:35 (PUBLIC)
[14:07:45][VV][esp32_ble_tracker:519]:   RSSI: -84
[14:07:45][VV][esp32_ble_tracker:520]:   Name: 'MiKetv12'
[14:07:45][VV][esp32_ble_tracker:528]:   Ad Flag: 6
[14:07:45][VV][esp32_ble_tracker:545]:   Service data:
[14:07:45][VV][esp32_ble_tracker:546]:     UUID: 0xFE95
[14:07:45][VV][esp32_ble_tracker:547]:     Data: 30.58.E0.09.03.35.C9.82.38.C1.A4.28.01.00 (14)
[14:07:45][VV][esp32_ble_tracker:550]: Adv data: 02.01.06.11.16.95.FE.30.58.E0.09.03.35.C9.82.38.C1.A4.28.01.00.09.09.4D.69.4B.65.74.76.31.32 (31)
```
I am pretty noob about all of this, so let me know if you need any info for me that would help you make this custom component compatible with this kettle.

tannisroot, 2022-07-22:
I know RSSI is pretty bad, but the ESP32 is only a meter away from the kettle, so not sure what is up with that.

dentra, 2022-07-22:
@tannisroot do you have any logs with your setup and starting with `miot` or `miot_kittle`?

tannisroot, 2022-07-22:
Sure, here, I redacted some sensitive info in WiFi section though. This is without verbose logging though.
[logs_esp32-board_upload.txt](https://github.com/dentra/esphome-components/files/9167815/logs_esp32-board_upload.txt)

After this the kettle entities just always show up as Unavailable

dentra, 2022-07-22:
Unfortunally still no `miot` or `miot_kittle` entries in the log. And even no logs from esp32_ble_tracker mapped to your device.

The advertisement from your first log is only tell some info about device and no object data

tannisroot, 2022-07-22:
Is that not it?
```[14:18:35][C][miot_kettle:243]: Xiaomi universal kettle
[14:18:35][C][miot_kettle:245]:   MAC: a4:c1:38:82:c9:35
[14:18:35][C][miot_kettle:247]:   Bindkey: 2C.93.ED.AC.A9.BB.A2.94.86.FC.25.E9.43.45.3D.A5 (16)
[14:18:35][C][miot_kettle:011]:   Water Boil Power 'Kettle Water Boil Power'
[14:18:35][C][miot_kettle:011]:     Device Class: 'power'
[14:18:35][C][miot_kettle:012]:  Water Boil Temperature 'Kettle Water Boil Temperature'
[14:18:35][C][miot_kettle:012]:    Device Class: 'temperature'
[14:18:35][C][miot_kettle:012]:    State Class: 'measurement'
[14:18:35][C][miot_kettle:012]:    Unit of Measurement: '°C'
[14:18:35][C][miot_kettle:012]:    Accuracy Decimals: 1
[14:18:35][C][esp32_ble_tracker:712]: BLE Tracker:
[14:18:35][C][esp32_ble_tracker:713]:   Scan Duration: 300 s
[14:18:35][C][esp32_ble_tracker:714]:   Scan Interval: 320.0 ms
[14:18:35][C][esp32_ble_tracker:715]:   Scan Window: 30.0 ms
[14:18:35][C][esp32_ble_tracker:716]:   Scan Type: ACTIVE
```

tannisroot, 2022-07-22:
I can record another log output similar to the previous one with verbose channel enabled if you need

tannisroot, 2022-07-22:
> After this the kettle entities just always show up as Unavailable

Sorry, derped out, I meant Unknown

dentra, 2022-07-22:
> Is that not it?
> 
> ```
> [14:18:35][C][miot_kettle:245]:   MAC: a4:c1:38:82:c9:35
> [14:18:35][C][miot_kettle:247]:   Bindkey: 2C.93.ED.AC.A9.BB.A2.94.86.FC.25.E9.43.45.3D.A5 (16)
> [14:18:35][C][miot_kettle:011]:   Water Boil Power 'Kettle Water Boil Power'
> [14:18:35][C][miot_kettle:011]:     Device Class: 'power'
> [14:18:35][C][miot_kettle:012]:  Water Boil Temperature 'Kettle Water Boil Temperature'
> [14:18:35][C][miot_kettle:012]:    Device Class: 'temperature'
> [14:18:35][C][miot_kettle:012]:    State Class: 'measurement'
> [14:18:35][C][miot_kettle:012]:    Unit of Measurement: '°C'
> [14:18:35][C][miot_kettle:012]:    Accuracy Decimals: 1
> [14:18:35][C][esp32_ble_tracker:712]: BLE Tracker:
> [14:18:35][C][esp32_ble_tracker:713]:   Scan Duration: 300 s
> [14:18:35][C][esp32_ble_tracker:714]:   Scan Interval: 320.0 ms
> [14:18:35][C][esp32_ble_tracker:715]:   Scan Window: 30.0 ms
> [14:18:35][C][esp32_ble_tracker:716]:   Scan Type: ACTIVE
> ```

it is not. it is just configuration information and not a device log. just wait a little, the advertisements from device goes in their own interval of course if the device support it.

dentra, 2022-07-22:
my integration will be able to work only if the device sends advertising packets about its status, if the device supports only the classic BT-connection, then unfortunately it will not work

tannisroot, 2022-07-22:
Oh, I've read somewhere that the device does indeed require classic BT connection. I was hoping this component supports that, since the kettle that is supported is practically the same mode, but apparently not.
Thanks for your work anyway!

closed 2022-07-22 by tannisroot

pattyland, 2022-07-25:
Hi, I have the same model and was about to report the same error... Is this now confirmed that this doesn't work or is this a guess?

If so, then would be nice to introduce next to the list of working devices also a list of incompatible devices