dentra/esphome-components #30: 2022.8.3

agentssmits, 2022-09-16:
I  got issue with integrating binary sensor miot_mccgq02hl:

 File ".esphome\external_components\d5822579\components\miot_mccgq02hl\binary_sensor.py", line 10, in <module>
    from .. import miot  # pylint: disable=relative-beyond-top-level
  File ".esphome\external_components\d5822579\components\miot\__init__.py", line 23, in <module>
    from .. import xiaomi_account  # pylint: disable=relative-beyond-top-level
  File ".esphome\external_components\d5822579\components\xiaomi_account\__init__.py", line 13, in <module>
    from .xiaomi_account import XiaomiAccount
  File ".esphome\external_components\d5822579\components\xiaomi_account\xiaomi_account.py", line 12, in <module>
    class XiaomiAccount:
  File ".esphome\external_components\d5822579\components\xiaomi_account\xiaomi_account.py", line 17, in XiaomiAccount
    servers: list[str],
TypeError: 'type' object is not subscriptable←[0m

(esphome 2022.8.3)

dentra, 2022-10-08:
Look like something wring with python environment. Please provide the config and environment in which you are building it. At this moment all components are tested with 2022.9.x and Python 3.9.2

agentssmits, 2022-10-12:
Thank you for giving me a pointer to update my Python from 3.8 to latest 3.10.2 - now it works again!

closed 2022-10-12 by agentssmits