dentra/esphome-components #31: XMWSDJ04MMC - No battery level

cvjensen, 2022-10-15:
Hi there , I have added this device in my esp config:

```
  - platform: miot_thermogigro
    mac_address: AA:BB:CC:DD:EE:FF
    bindkey: "1234567...."
    name: "E-ink Test Temperatur"
    humidity:
      name: "E-ink Test Fugtighed"
    battery_level:
      name: "E-ink Test Batteri"

  - platform: ble_rssi
    mac_address: AA:BB:CC:DD:EE:FF
    name: "E-ink Test signalstyrke"
    id: test_rssi
```

Everything seems fine except the battery data:

[image]

How can that be ?

dentra, 2022-10-16:
Hi @cvjensen! Sometimes it may take several hours before the battery status changes.

Also if you have “Unhandled object attribute” in the log, please let me take a look at it.

cvjensen, 2022-10-18:
It has now been running for 3 days and battery still shows as unknown .

Which log should I look in ? just the system log? Cause there are no “Unhandled object attribute”

dentra, 2022-10-21:
I currently support two types of battery charge attributes, standard and XMWSDJ04MMC-specific. In your case, for some reason, the device does not send data or sends something completely different, which is why the message "Unhandled object attribute" is interesting for me. We are also interested in messages related to the sensor "E-ink Test Batteri", there may be an error in the code.

cvjensen, 2022-10-24:
I had some troubles with my esp suddenly starting to reboot all the time so I had to remove the new XMWSDJ04MMC and your code . I dont know if its a lack of processing power, errors in the code or just too many BT devices connected . 
I have ordered a new stronger ESP32 and will test it later on . Then we can take a look at your code and my BT device together.

cvjensen, 2022-11-11:
hi again , I'm now running on a test esp32 with only the new XMWSDJ04MMC connected . 
can you guide me through what you need to debug? 
Just to make it clear it still doesnt show up:

[image]

cvjensen, 2022-11-14:
I just tried to take out the battery and put it back in and now:

[image]

So I'm guessing the firmware really really dont update the battery information unless it changes ?

dentra, 2022-11-14:
Hi @cvjensen yes, my integration is passive, it could update data only when it sent by device.

closed 2023-01-20 by stale[bot]