dentra/esphome-components #32: Can't get Mi Smart Kettle Pro working

staxDB, 2022-10-22:
Hey there, I've tried to integrate the [Mi Smart Kettle Pro](https://www.mi.com/de/product/mi-smart-kettle-pro/) using [this](https://github.com/dentra/esphome-components/tree/master/components/miot_kettle) documentation with my esphome, but the state of the sensors are still `unknown`

Here is my configuration:

```yaml
external_components:
  - source: github://dentra/esphome-components

binary_sensor:
  - platform: miot_kettle
    # String (Required), device MAC-address.
    mac_address: A4:C1:38:8B:72:7F
    # String, (Optional), device bind key. Will use "xiaomi_account" from "miot" if absent to automaticaly get the bindkey.
    bindkey: "5F3*************"
    # String, (Optional), the name of binary sensor
    name: "Kettle Water Boil Power"
    # Sensor (Optional), Temperature, °C
    temperature:
      name: "Kettle Water Boil Temperature"
```

Here is the log output:

```
[00:46:20][C][miot_kettle:243]: Xiaomi universal kettle
[00:46:20][C][miot_kettle:245]:   MAC: a4:c1:38:8b:72:7f
[00:46:20][C][miot_kettle:247]:   Bindkey: 5F.**.**.**.**.**.**.**.**.**.**.**.**.**.**.** (16)
[00:46:20][C][miot_kettle:011]:   Water Boil Power 'Kettle Water Boil Power'
[00:46:20][C][miot_kettle:011]:     Device Class: 'power'
[00:46:20][C][miot_kettle:012]:  Water Boil Temperature 'Kettle Water Boil Temperature'
[00:46:20][C][miot_kettle:012]:    Device Class: 'temperature'
[00:46:20][C][miot_kettle:012]:    Accuracy Decimals: 1
....
[00:46:20][D][miot.automation:009]:   Product ID: 09E0
[00:46:20][D][miot.automation:010]:   Frame Control:
[00:46:20][D][miot.automation:011]:     is_encrypted      : NO
[00:46:20][D][miot.automation:012]:     mac_include       : YES
[00:46:20][D][miot.automation:013]:     capability_include: YES
[00:46:20][D][miot.automation:014]:     object_include    : NO
[00:46:20][D][miot.automation:015]:     mesh              : NO
[00:46:20][D][miot.automation:016]:     registered        : NO
[00:46:20][D][miot.automation:017]:     solicited         : NO
[00:46:20][D][miot.automation:018]:     auth_mode         : 2
[00:46:20][D][miot.automation:019]:     version           : 5
[00:46:20][D][miot.automation:020]:   Frame Counter: 1
[00:46:20][D][miot.automation:023]:   MAC address: a4:c1:38:8b:72:7f
[00:46:20][D][miot.automation:026]:   Capability:
[00:46:20][D][miot.automation:027]:     connectable : NO
[00:46:20][D][miot.automation:028]:     centralable : NO
[00:46:20][D][miot.automation:029]:     encryptable : NO
[00:46:20][D][miot.automation:030]:     bond_ability: 1
[00:46:20][D][miot.automation:031]:     io          : YES
[00:46:20][D][miot.automation:035]:     can_enter_digits     : YES
[00:46:20][D][miot.automation:036]:     can_enter_letters    : NO
[00:46:20][D][miot.automation:037]:     can_read_nfc_tags    : NO
[00:46:20][D][miot.automation:038]:     can_recognize_qr_code: NO
[00:46:20][D][miot.automation:039]:     can_output_digits    : NO
[00:46:20][D][miot.automation:040]:     can_output_letters   : NO
[00:46:20][D][miot.automation:041]:     can_generate_nfc_tag : NO
[00:46:20][D][miot.automation:042]:     can_generate_qr_code : NO
```

And here are screenshots of the sensors states:
[image]
[image]

What am I missing here?

znack14, 2022-11-29:
Доброй ночи!!!
У меня такая же проблема с этим чайником.
И bindkey и все остальное, но не работает

dentra, 2022-12-09:
my integration will be able to work only if the device sends advertising packets about its status, if the device supports only the classic BT-connection, then unfortunately it will not work.

dentra, 2022-12-18:
@xrust83 Unfortunately it is technically impossible with passive BLE.

closed 2023-02-24 by stale[bot]