dentra/esphome-components #34: platform: miot_mccgq02hl

andrewbeyou88, 2022-12-08:
Hi, i am using "ESPHome Current version: 2022.11.5". I just updated and i was looking to integrate one of Xiaomi mccgq02hl sensors in my ESP32 bluetooth tracker.
     I am not sure if i am doying it wrong or is some incompatibility issue. With this ESP32 i have succesfully integrated "platform: atc_mithermometer" sensors, yet i have these door sensors that i never managed to integrate. 
    The error i got is: "Failed config
binary_sensor.miot_mccgq02hl: [source /config/esphome/bt-scanner.yaml:90]"

This is my ESP configuration:

bt-scanner.yaml
"
esphome:
  name: bt-scanner
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "#####"
  password: !secret wifi_password
  manual_ip:
    static_ip: ###
    gateway: ###
    subnet: ###

captive_portal:

# Enable logging
logger:
  level: debug
# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: "####"

esp32_ble_tracker:
  scan_parameters:
    interval: 15000ms
    window: 1500ms
    active: true

bluetooth_proxy:
  active: true

external_components:
  - source: github://dentra/esphome-components

switch:
  - platform: gpio
    name: "ESP32-Onboard-LED"
    pin: 2

  - platform: restart
    name: ESP32-RestartSW
    id: restart_switch

sensor:
  - platform: uptime
    name: "ESP32_Uptime Sensor"
    
  - platform: wifi_signal
    name: "ESP32 WiFi Signal"
    update_interval: 60s
    
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:81:68:2B"
    temperature:
      name: "Sensor1 Temperature"
    humidity:
      name: "Sensor1 Humidity"
    battery_level:
      name: "Sensor1 Battery Level"    

  - platform: atc_mithermometer
    mac_address: "A4:C1:38:4C:02:B2"
    temperature:
      name: "Sensor2 Temperature"
    humidity:
      name: "Sensor2 Humidity"
    battery_level:
      name: "Sensor2 Battery Level"
        
  - platform: atc_mithermometer
    mac_address: "A4:C1:38:FA:32:0F"
    temperature:
      name: "Sensor3 Temperature"
    humidity:
      name: "Sensor3 Humidity"
    battery_level:
      name: "Sensor3 Battery Level"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
    name: "door locked"
    device_class: door

  - platform: miot_mccgq02hl
    name: mccgq02hl
    mac_address: E4:AA:EC:36:D9:3A
    bindkey: "####"
    battery_level:
      name: "Door sensor battery level"
    light:
      name: "Door sensor light"
    opening:
      name: "Door sensor open"
"

Please help me!
Sincerely, Andrei Dumitru

andrewbeyou88, 2022-12-08:
[opening] is an invalid option for [binary_sensor.miot_mccgq02hl]. Please check the indentation.
  opening: 
    name: Door sensor open

andrewbeyou88, 2022-12-08:
DONE! 
i erase
"opening:
name: "Door sensor open"  
and just added 
"device_class: opening"

- platform: miot_mccgq02hl
    name: mccgq02hl
    mac_address: E4:AA:EC:36:D9:3A
    bindkey: "####"
    battery_level:
      name: "Door sensor battery level"
    light:
      name: "Door sensor light"
   device_class: opening

closed 2022-12-08 by andrewbeyou88

dentra, 2022-12-09:
My component do not support opening configuration option. You can look documentation for all additional options here https://github.com/dentra/esphome-components/tree/master/components/miot_mccgq02hl

andrewbeyou88, 2022-12-09:
Thank you for your answer. I managed to fix my mistake, as i commented at
the end.