dentra/esphome-components #35: LYWSD02MMC is not working

antonioag95, 2023-01-06:
Despite providing the bindkey packets from LYWSD02MMC cannot be decoded:

```
[19:04:51][W][miot:193]: A4C138693D61 [16E4] Unhandled object attribute: 4C02, value: 43
[19:04:51][D][miot:146]: Got MiBeacon: 58.58.E4.16.7E.64.3D.89.38.C1.A4.23.B3.43.C1.7F.84.00.F1.63.64.07 (22)
[19:04:51][D][miot:149]:    [16E4] (encrypted) A4:C1:38:69:3D:61 RSSI=-62 ▂▄▆█
[19:04:55][W][miot:193]: A4C138693D61 [16E4] Unhandled object attribute: 4C02, value: 43
[19:04:55][D][miot:146]: Got MiBeacon: 58.58.E4.16.82.64.3D.89.38.C1.A4.F5.FF.24.A0.7F.84.00.9C.89.82.DB (22)
[19:04:55][D][miot:149]:    [16E4] (encrypted) A4:C1:38:69:3D:61 RSSI=-75 ▂▄▆█
[19:05:00][W][miot:193]: A4C138693D61 [16E4] Unhandled object attribute: 4C02, value: 43
[19:05:00][D][miot:146]: Got MiBeacon: 58.58.E4.16.86.64.3D.89.38.C1.A4.18.44.1B.76.7F.84.00.3C.42.41.43 (22)
[19:05:00][D][miot:149]:    [16E4] (encrypted) A4:C1:38:69:3D:61 RSSI=-61 ▂▄▆█
[19:05:02][W][miot:193]: A4C138693D61 [16E4] Unhandled object attribute: 4C02, value: 43
[19:05:02][D][miot:146]: Got MiBeacon: 58.58.E4.16.88.64.3D.89.38.C1.A4.6C.F9.AD.06.7F.84.00.D8.A5.8A.7D (22)
[19:05:03][D][miot:149]:    [16E4] (encrypted) A4:C1:38:69:3D:61 RSSI=-74 ▂▄▆█
[19:05:05][W][miot:193]: A4C138693D61 [16E4] Unhandled object attribute: 4C02, value: 43
[19:05:05][D][miot:146]: Got MiBeacon: 58.58.E4.16.8A.64.3D.89.38.C1.A4.28.A9.1E.81.7F.84.00.C9.2E.3D.EC (22)
[19:05:05][D][miot:149]:    [16E4] (encrypted) A4:C1:38:69:3D:61 RSSI=-62 ▂▄▆█
```

Can you please help with that? Thanks!

antonioag95, 2023-01-12:
Thanks for your answer @dentra I see that, thanks for clarifying. However I am not aware of what value 43 is, I guess it might be either the temperature, humidity or battery? I found another log on the internet about the same device:

```
[14:08:26][W][miot:193]: A4C138D5E665 [16E4] Unhandled object attribute: 4C02, value: 2F
[14:08:26][D][miot:146]: Got MiBeacon: 58.58.E4.16.AF.65.E6.D5.38.C1.A4.49.94.2C.DD.E1.01.00.84.33.42.57 (22) [14:08:26][D][miot:149]: [16E4] (encrypted) A4:C1:38:D5:E6:65 RSSI=-54 ▂▄▆█
```

dentra, 2023-01-12:
Do you have any Temperature, RH or Battery level for your device?

Sorry, I am forgot that unhandled value is a dump, so 43 is one byte in hex - in your sample it is 67 in decimal. You could look at display when the log show the value.

antonioag95, 2023-01-12:
Yes! It is RH! I'm leaving it decoding packets for some hours until we try to figure out what all values are referring to!

So for now we know that object attribute 4C02 is the RH.

dentra, 2023-01-12:
Hm… possible it is a fresh version with new attributes. I will glad to support it.

antonioag95, 2023-01-12:
Thank you for that! The weird part is that it seems to be sending RH only for now. I'll log everything overnight to see if something else appears. I'll keep you updated!

dentra, 2023-01-12:
The temperature and battery level of your device are also not displayed?

Do you know the model of our device? Most likely it starts with miaomiaoce.sensor_ht.?

antonioag95, 2023-01-12:
Nothing is displayed but I cannot see anything related to either the temperature or the battery in the logs. Whatever I see is referred to the RH.

Sure, the model name is _miaomiaoce.sensor_ht.o2_

dentra, 2023-01-12:
I'm add new RH attribute asap, to stop you device flooding it.

unfortunately miot-spec only shown already supported attributes for your model :(

dentra, 2023-01-13:
updated. also i disable additional checks for miaomiaoce temperature and battery, so it may work now

antonioag95, 2023-01-13:
Thank you very much @dentra everything is working now!
[image]

closed 2023-01-13 by antonioag95