dentra/esphome-components #39: add weekday to energy tariffs

wazoo666, 2023-06-11:
Hi
It is possible to add "weekday"  and "holiday" to tariffs component ??  :)

I mean, actually in Spain PVPC has 3 tariff , like P1 for peak (expensive), P2 for mid and P3 for valley (cheap)
but they are separated in weekday too
I mean, P1, P2 and P3 are used for working days (mon-fri),  buy (sat-sun) are 24h all same P3 tariff
and to complicate this situation it adds national holidays that always are P3  (12 days over the year)
ThankU

dentra, 2023-06-27:
do you mean something like this:
```yaml
...
      - name: "P1"
        time: "06:00-14:00"
        days_of_week: &workdays MON-FRI
        exclude: &holidays [25.12, 26.12, 27.12]
      - name: "P2"
        time: "14:00-22:00"
        days_of_week: *workdays
        exclude: *holidays
      - name: "P3"
...
```

wazoo666, 2024-06-15:
> do you mean something like this:
> 
------------------------------
sorry for my delayed readout.
I think yes, will test it
thankU

dentra, 2026-10-13:
Implemented in 37a7c6f, available in `master`.

New options:
- `days_of_week` per tariff: `MON-FRI`, `SAT, SUN`, `[SAT, SUN]`, `FRI-MON`, numbers `1-7` (1 is Sunday). Without `time`, the tariff is active all day.
- `holidays` at component level: quoted `"DD.MM"` (every year) or `"YYYY-MM-DD"` (exact date). On a holiday the day of week is replaced with `holiday_as` (default `SUN`).
- `holiday`: optional binary sensor (e.g. HA Workday) for holidays that aren't on a fixed date.

PVPC 2.0TD example: P1/P2 on workdays; P3 at night, at weekends and on national holidays (P3 is the default tariff):

```yaml
sensor:
  - platform: energy_tariffs
    total: total
    holidays: ["01.01", "06.01", "01.05", "15.08", "12.10", "01.11", "06.12", "08.12", "25.12"]
    tariffs:
      - name: P1
        time: ["10:00-14:00", "18:00-22:00"]
        days_of_week: MON-FRI
      - name: P2
        time: ["08:00-10:00", "14:00-18:00", "22:00-00:00"]
        days_of_week: MON-FRI
      - name: P3
```

Details are in the [README](https://github.com/dentra/esphome-components/tree/master/components/energy_tariffs#days-of-week-and-holidays). If you find a problem or have a suggestion, please open a new issue.