dentra/esphome-components #44: Manual EraClean Intelligent Deodorizer Max

DivanX10, 2023-08-12:
Thanks for this component. EraClean Intelligent Deodorizer Max is perfectly controlled and the sensors work. I want to add for reference, so that there is an understanding if someone also receives the Failed login: 0x0023 error. I ran into a problem and here I did not find an answer. Then I realized the reason why the Failed login error occurred: 0x0023. EraClean Intelligent Deodorizer Max does not support multiple connections and if the phone where MiHome is installed and bluetooth is enabled, then EraClean Intelligent Deodorizer Max will connect to the phone by default and esp32 will lose contact with EraClean Intelligent Deodorizer Max. In the logs, ESPHome will receive the Failed login error: 0x0023

Enable logging
```
logger:
  level: DEBUG
```

Log with connection error, where we see Failed login: 0x0023
```
[12:15:20][D][esp32_ble_client:048]: [0] [XX:XX:XX:XX:XX:XX] Found device
[12:15:20][D][esp32_ble_tracker:213]: Pausing scan to make connection...
[12:15:20][I][esp32_ble_client:064]: [0] [XX:XX:XX:XX:XX:XX] 0x00 Attempting BLE connection
[12:15:23][I][esp32_ble_client:196]: [0] [XX:XX:XX:XX:XX:XX] Connected
[12:15:23][D][esp32_ble_tracker:245]: Starting scan...
[12:15:23][D][miot_standard_auth_client:090]: state=0xB (11)
[12:15:24][D][miot_standard_auth_client:090]: state=0xB (11)
[12:15:24][D][miot_standard_auth_client:090]: state=0x0 (0)
[12:15:24][D][miot_standard_auth_client:090]: state=0xD (13)
[12:15:24][D][miot_standard_auth_client:090]: state=0x0 (0)
[12:15:24][D][miot_standard_auth_client:090]: state=0xC (12)
[12:15:24][D][miot_standard_auth_client:090]: state=0xC (12)
[12:15:24][W][miot_standard_auth_client:212]: Fail compare
[12:15:24][D][miot_standard_auth_client:090]: state=0xA (10)
[12:15:24][D][miot_standard_auth_client:090]: state=0xA (10)
[12:15:24][W][miot_standard_auth_client:069]: Failed login: 0x0023
[12:20:07][I][ota:113]: Boot seems successful, resetting boot loop counter.

```

Whatever happens, there is Bluetooth support on the phone with the MiHome connected, and the esp 32 can easily establish a connection with the EraClean Max intelligent deodorant.

```
14:24:12	[D]	[esp32_ble_client:048]	
[0] [XX:XX:XX:XX:XX:XX] Found device
14:24:12	[D]	[esp32_ble_tracker:213]	
Pausing scan to make connection...
14:24:12	[I]	[esp32_ble_client:064]	
[0] [XX:XX:XX:XX:XX:XX] 0x00 Attempting BLE connection
14:24:14	[I]	[esp32_ble_client:196]	
[0] [XX:XX:XX:XX:XX:XX] Connected
14:24:14	[D]	[esp32_ble_tracker:245]	
Starting scan...
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xB (11)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xB (11)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0x0 (0)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0x0 (0)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xD (13)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0x0 (0)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xC (12)
14:24:14	[D]	[miot_standard_auth_client:090]	
state=0xC (12)
14:24:15	[D]	[miot_standard_auth_client:090]	
state=0xA (10)
14:24:15	[D]	[miot_standard_auth_client:090]	
state=0xA (10)
14:24:15	[I]	[miot_standard_auth_client:062]	
Login Success
```

DivanX10, 2023-08-13:
**Below is an example of a card and automation for Home Assistant**

The essence of automation is that you can switch modes at different times. There is one mode during the day, and another mode at night. Let's say the freshness mode is an ozonizer and I use it during the day, when we often open the refrigerator door and eliminate germs and odors, and at night, when everyone is asleep, the ionizer mode turns on and saturates the refrigerator chamber with negative ions all night. Freshness mode is the same deodorization mode, only with less work time, freshness has 3 minutes of work and 90 minutes of sleep, and deodorization has 15 minutes of work and 90 minutes of sleep. During the freshness or deodorization mode, electrical discharges are visible through the grate and the smell of ozone is very well felt, but in the ionizer mode there are no electrical discharges, only negative ions are produced.
 
**Electrical discharges when we turn on the freshness and deodorization mode**
[image]

**A card for controlling the ionizer. Card view in collapsed form**
[image]

**A card for controlling the ionizer. Card view in expanded form**
[image]

<details>
  <summary>Card</summary>

```
type: entities
entities:
  - type: custom:fold-entity-row
    head:
      entity: sensor.kholodilnik_lg_ionizator
      name: Ionizer
      icon: mdi:air-filter
      secondary_info:
        attribute: battery
        name: Battery
        unit: '%'
      type: custom:multiple-entity-row
      show_state: false
      state_header: Status
      entities:
        - entity: group.eraclean_airpurifier_menu
          name: Menu
          state_color: true
          icon: mdi:information-outline
          styles:
            height: 60px
            width: 50px
        - entity: switch.eraclean_airpurifier_power
          name: Ionizer
          type: button
          state_color: true
          icon: mdi:creation
          tap_action:
            action: toggle
          styles:
            height: 60px
            width: 50px
        - entity: switch.eraclean_airpurifier_cycle
          name: Cycle
          type: button
          state_color: true
          icon: mdi:backup-restore
          tap_action:
            action: toggle
          styles:
            height: 60px
            width: 50px
        - entity: select.eraclean_airpurifier_mode
          name: Mode
          styles:
            height: 50px
            width: 50px
    entities:
      - entity: input_boolean.eraclean_airpurifier_auto_mode
        name: Auto Mode
      - entity: input_datetime.eraclean_airpurifier_daytime
        name: Daytime
      - entity: input_select.eraclean_airpurifier_day_mode
[... 151 lines cut ...]
            target:
              entity_id: select.eraclean_airpurifier_mode
            alias: Set the day mode
      - conditions:
          - condition: time
            after: input_datetime.eraclean_airpurifier_nighttime
            before: input_datetime.eraclean_airpurifier_daytime
            alias: Night time
          - condition: template
            value_template: >-
              {{ states('input_datetime.eraclean_airpurifier_nighttime') !=
              states('select.eraclean_airpurifier_mode') }}
            alias: Night mode is not equal to the set mode
        sequence:
          - service: select.select_option
            data:
              option: "{{ states('input_select.eraclean_airpurifier_night_mode') }}"
            target:
              entity_id: select.eraclean_airpurifier_mode
            alias: Set night mode
mode: single

```
  
</details>

closed 2023-10-20 by stale[bot]

dentra, 2026-10-13:
Thanks @DivanX10 for the detailed investigation! I've added the `Failed login: 0x0023` explanation to the troubleshooting section of the README and linked this issue as a Home Assistant example.