dentra/esphome-components #45: miot_mccgq02hl error won't compile "collect2: error: ld returned 1 exit status"

JakeTheRabbit, 2023-08-21:
Hi, 

This is my yaml config: 

`esphome:
  name: esppoe-4
  friendly_name: esppoe-4
  name_add_mac_suffix: true
  project:
    name: esphome.bluetooth-proxy
    version: "1.0"

esp32:
  board: m5stack-core-esp32
  framework:
    type: esp-idf

# Enable logging
logger:

# Keep this for remote flashing capability
ota:

# Network interface for PoESP32
ethernet:
  type: IP101
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO5

api:
  encryption:
    key: "kwgXOnw6i3lnKGVB+4AkA2YqmRSCrY3gi2SomEDE1lo="

external_components:
  - source: github://dentra/esphome-components

binary_sensor:
  - platform: gpio
    pin:
      number: 17
      inverted: true
    name: "BlueButton Door"
    id: "BlueButton"
    on_press:
      - logger.log: "Blue button door pressed"
  - platform: gpio
    pin: 
      number: 16
      inverted: true
    name: "RedButton Door"
    id: "RedButton"
    on_press:
      - logger.log: "Red button door pressed"

  - platform: miot_mccgq02hl
    # String (Required), device MAC-address.
    mac_address: "E4:AA:EC:71:D4:27"
    # String, (Optional), device bind key. Will use "xiaomi_account" from "miot" if absent to automaticaly get the bindkey.
    bindkey: "4bee582512114112c956f122f741da59"
     # String, (Optional), the name of binary sensor
    name: "$name Door/Window Sensor"
    # BinarySensor (Optional), Light intensivity: on - strong light, off - weak light
    light:
      name: "$name Light"
    # BinarySensor (Optional), Alert enabled when door/window has opened for a long (device configured) time
    alert:
      name: "$name Alert"
[... 54 lines cut ...]
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o:(.literal._Z5setupv+0x124): undefined reference to `vtable for esphome::miot::MiBeaconTracker'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o:(.literal._Z5setupv+0x1d8): undefined reference to `esphome::sensor::Sensor::Sensor()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o:(.literal._Z5setupv+0x1dc): undefined reference to `esphome::sensor::Sensor::set_state_class(esphome::sensor::StateClass)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o:(.literal._Z5setupv+0x1e4): undefined reference to `esphome::sensor::Sensor::set_accuracy_decimals(signed char)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o: in function `setup()':
/config/esphome/.esphome/build/esppoe-4/src/main.cpp:359: undefined reference to `esphome::sensor::Sensor::Sensor()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o: in function `__gnu_cxx::new_allocator<esphome::esp32_ble::BLEEvent**>::allocate(unsigned int, void const*)':
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/ext/new_allocator.h:111: undefined reference to `esphome::sensor::Sensor::set_state_class(esphome::sensor::StateClass)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o: in function `std::_Deque_iterator<esphome::esp32_ble::BLEEvent*, esphome::esp32_ble::BLEEvent*&, esphome::esp32_ble::BLEEvent**>::_M_set_node(esphome::esp32_ble::BLEEvent***)':
/data/cache/platformio/packages/toolchain-xtensa-esp32/xtensa-esp32-elf/include/c++/8.4.0/bits/stl_deque.h:259: undefined reference to `esphome::sensor::Sensor::set_accuracy_decimals(signed char)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o: in function `esphome::esp32_ble::Queue<esphome::esp32_ble::BLEEvent>::Queue()':
/config/esphome/.esphome/build/esppoe-4/src/esphome/components/esp32_ble/queue.h:24: undefined reference to `esphome::sensor::Sensor::Sensor()'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o: in function `esphome::miot::MiotListener::set_address(unsigned long long)':
/config/esphome/.esphome/build/esppoe-4/src/esphome/components/miot/miot.h:67: undefined reference to `esphome::sensor::Sensor::set_state_class(esphome::sensor::StateClass)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o: in function `setup()':
/config/esphome/.esphome/build/esppoe-4/src/main.cpp:373: undefined reference to `esphome::sensor::Sensor::set_accuracy_decimals(signed char)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/main.o:(.rodata._ZTVN7esphome4miot13MiotComponentE[vtable for esphome::miot::MiotComponent]+0x40): undefined reference to `esphome::miot::MiotListener::process_mibeacon(esphome::miot::MiBeacon const&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/esphome/core/controller.o:(.literal._ZN7esphome10Controller16setup_controllerEb+0x10): undefined reference to `esphome::sensor::Sensor::add_on_state_callback(std::function<void (float)>&&)'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /data/esppoe-4/.pioenvs/esppoe-4/src/esphome/core/controller.o: in function `esphome::Controller::setup_controller(bool)':
/config/esphome/.esphome/build/esppoe-4/src/esphome/core/controller.cpp:29: undefined reference to `esphome::sensor::Sensor::add_on_state_callback(std::function<void (float)>&&)'
collect2: error: ld returned 1 exit status
*** [/data/esppoe-4/.pioenvs/esppoe-4/firmware.elf] Error 1
========================= [FAILED] Took 21.30 seconds =========================`

Any ideas what I'm doing wrong?

dentra, 2023-08-21:
Hi, first try to clean and compile again

dentra, 2023-08-21:
I will check with ESPHome 2023.8.1

JakeTheRabbit, 2023-08-21:
> Hi, first try to clean and compile again

This worked lol thanks

closed 2023-08-31 by dentra