dentra/esphome-components #46: Accuracy of Energy Statistics component?

poldim, 2023-09-18:
Hey,
I've added your component to my Sonoff S31 and see the data is pretty different from the native component.  Have you done a similar check?  Wondering if it's my config, device, or one of the components? The data from your component averages being 55% of the native component.

Comparison of daily max for `garage_controls_s31_energy_today` vs `garage_controls_s31_total_daily_energy_native_esphome_component`. 
[image]

| date    | dentra  | native | delta |
|---------|---------|--------|-------|
| 9/7/23  | 0.67    | 1.3    | 52%   |
| 9/8/23  | 0.4     | 0.705  | 57%   |
| 9/9/23  | 0.37    | 0.72   | 51%   |
| 9/10/23 | 0.29    | 0.547  | 53%   |
| 9/11/23 | 0.66    | 1.21   | 55%   |
| 9/12/23 | 0.46    | 0.875  | 53%   |
| 9/13/23 | 0.36    | 0.688  | 52%   |
| 9/14/23 | 0.75    | 1.43   | 52%   |
| 9/15/23 | 0.29    | 0.406  | 71%   |
| 9/16/23 | 0.37    | 0.648  | 57%   |
| 9/17/23 | 0.11    | 0.211  | 52%   |
|         |         |        | 55%   |

My code:
```yaml
external_components:
- source: github://dentra/esphome-components

sensor:
- platform: cse7766
  update_interval: 10s
  current:
    name: "${friendly_name} Current"
  voltage:
    name: "${friendly_name} Voltage"
  power:
    name: "${friendly_name} Power"
    id: s31_power
    unit_of_measurement: "W"
    filters:
      - delta: 3.0
    on_value:
      then:
        - sensor.template.publish:
            id: s31_power_kw
            state: !lambda '{ return x; }'
- platform: template
  name: "${friendly_name} in kW"
  id: s31_power_kw
  internal: true
  filters:
    - delta: 5
    - multiply: 0.001
  unit_of_measurement: "kW"
  accuracy_decimals: 3

- platform: total_daily_energy
  name: "${friendly_name} Total Daily Energy (Native ESPHome Component)"
  power_id: s31_power_kw

- platform: integration
  name: "${friendly_name} Total Energy"
  id: total_energy_by_integration
  sensor: s31_power
  time_unit: h
  unit_of_measurement: "kWh"
  filters:
    - multiply: 0.001
  accuracy_decimals: 3
  restore: true
  state_class: total_increasing

- platform: "energy_statistics"
  total: total_energy_by_integration
  energy_today:
    name: "${friendly_name} Energy Today"
  energy_yesterday:
    name: "${friendly_name} Energy Yesterday"
  energy_week:
    name: "${friendly_name} Energy Week"
  energy_month:
    name: "${friendly_name} Energy Month"
```

dentra, 2023-11-24:
My component has been actively tested and actually works together with the zmai-90; there are practically no deviations in readings from the main meter over several years of use.

dentra, 2023-11-24:
you calculate the data differently so you see different results

closed 2024-02-06 by poldim