dentra/esphome-components #50: Unable to load component partitions

backinside, 2023-11-30:
Hi,

I'm having trouble adding coredump and partitions to my project
Am I missing something?

ERROR Unable to load component partitions:
Traceback (most recent call last):
  File "C:\Users\Peter\AppData\Local\Programs\Python\Python311\Lib\site-packages\esphome\loader.py", line 165, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Peter\AppData\Local\Programs\Python\Python311\Lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "c:\Work\nspanel\test1\.esphome\external_components\2f2a5be3\components\partitions\__init__.py", line 8, in <module>
    from .const import *
  File "c:\Work\nspanel\test1\.esphome\external_components\2f2a5be3\components\partitions\const.py", line 3, in <module>
    DOMAIN = esp32.CONF_PARTITIONS
             ^^^^^^^^^^^^^^^^^^^^^
AttributeError: module 'esphome.components.esp32' has no attribute 'CONF_PARTITIONS'
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
Failed config

partitions: [source emeletgyerek2.yaml:9]

  Component not found: partitions.
  coredump:

dentra, 2023-12-07:
What version of EspHome are you using?

backinside, 2023-12-07:
oh I missed that, I'll update :)

dentra, 2023-12-08:
Do you solve it?