dentra/esphome-components #51: Xiaomi Mijia Night Light 2 (MJYD02YL) - longer detection time?

rwjack, 2023-12-01:
Hey, could these components help with having the light detect movement for longer? 

Even though I configured the light to stay awake for 60 seconds in the Mi app, it keeps randomly turning off, even when there is movement.

dentra, 2023-12-07:
Could you please share logs?

rwjack, 2023-12-08:
Just to clarify, I haven't configured anything yet because I'm still waiting on an esp32 to use as a BT proxy, but I'm just investigating if this is worth implementing for my use case.

rwjack, 2023-12-08:
I think this is what would work: https://esphome.io/components/bluetooth_proxy.html

Just to confirm, if the proxy can forward BLE advertisements to HA, that means I can set the light activation time to longer than 60s via this component?

Although I cannot turn the light on and off manually?

rwjack, 2024-01-01:
I got the BT proxy up and running, and connected to the lights using: https://github.com/al-one/hass-xiaomi-miot

Although the integration doesn't expose an option to toggle the lights on or off.

Can your component do so?

dentra, 2024-01-12:
I bought this device and research it.

ov1d1u, 2024-01-29:
Any updates, @dentra? I also looked into this and I see that there are some values written on the `00000101-0065-6c62-2e74-6f696d2e696d` characteristic when turning the light on or off:

```
OFF: 0d00891914aed72db5b0
ON:  0e006d7f8326c7edae48
```

and in another session:
```
OFF: 0d00c0220da32b2e2b8f
ON:  0e004c308167a88e0fba
```

It seems that the first byte is some kind of counter, the second is always `00`, but the rest differs a lot which makes me think there is some kind of encryption/scrambling.

dentra, 2024-02-03:
It is not possible to control on/off manually. 
It is only possibly to change interval of reporting via MiHome.

ov1d1u, 2024-02-05:
@dentra You can control the night light mode through the app, when you are directly connected to the light:

[image]

closed 2024-04-18 by stale[bot]