dentra/esphome-components #53: Bakcup broken since ESPHome 2023.12

nagyrobi, 2023-12-14:
The contents of the backed up file is:

```
null
...

```

Also, the GPIO pin validation fails with this component added:

```
INFO ESPHome 2023.12.0b1
INFO Reading configuration /root/config/kazan-124.yaml...
INFO Generating C++ source...
INFO Reading configuration /root/config/kazan-124.yaml...
Failed config

binary_sensor.gpio: [source <unicode string>:157]
  
  Pin 5 is used in multiple places.
  platform: gpio
  pin: 
    number: 5
    inverted: True
    mode: 
      input: True
      pullup: True
      output: False
      open_drain: False
      pulldown: False
      analog: False
  name: Junkers betáp
  device_class: power
  id: power_active
  filters: 
    - delayed_on: 300ms
    - delayed_off: 300ms
  disabled_by_default: False
light.status_led: [source <unicode string>:183]
  
  Pin 2 is used in multiple places.
  platform: status_led
  id: onboard_led
  internal: True
  pin: 
    number: 2
    inverted: True
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
  disabled_by_default: False
  restore_mode: ALWAYS_OFF
  name: onboard_led
output.esp8266_pwm: [source <unicode string>:240]
  
  Pin 15 is used in multiple places.
  platform: esp8266_pwm
  id: pwm_output_hw
  pin: 
    number: 15
    mode: 
      output: True
      input: False
      open_drain: False
      pullup: False
      pulldown: False
      analog: False
    inverted: False
  frequency: 1000.0
  zero_means_zero: False
Pin 5 is used in multiple places

Pin 2 is used in multiple places

Pin 15 is used in multiple places

INFO Backup config will take: 29 bytes
INFO Compiling app...
```

Reference Discord: https://discord.com/channels/429907082951524364/1184783083048812616

mesistostefre, 2023-12-21:
true ... and "allow_other_uses: true" does not resolve the issue

dentra, 2024-01-12:
Thanks for reporting. I will investigate it.

dentra, 2024-01-15:
fixed

closed 2024-01-16 by dentra