dentra/esphome-components #57: Config bindkey rather than providing Mi account for miot_cwbs01

keniji, 2024-04-08:
First thanks for these wonderful integrations, especially these on the MIOT platform, they enable the possibility of controlling IoT devices.
I have recently bought a CWBS-01 and am using [your component](https://github.com/dentra/esphome-components/tree/master/components/miot_cwbs01), may I know if it's possible to config `bindkey` for this component, rather than providing `xiaomi_account`? I've tried this config (like [miot_ylkg0xyl](https://github.com/dentra/esphome-components/tree/master/components/miot_ylkg0xyl)):
```
binary_sensor:
  - platform: miot_cwbs01
    ble_client_id: ble_client_deodorizer
    bindkey: "$device_bindkey"
```
But no matther I use `BLE KEY` or `TOKEN`, which are from [Xiaomi-cloud-tokens-extractor](https://github.com/PiotrMachowski/Xiaomi-cloud-tokens-extractor), I always get errors:
```
[20:45:59][W][miot_standard_auth_client:212]: Fail compare
...
[20:45:59][W][miot_standard_auth_client:069]: Failed login: 0x0023
```
I guess it's because the key is not accepted?
Thanks in advance!

dentra, 2024-05-12:
For this integration you need to configure both - bindkey and token.
xiaomi_account is a helper that works only on code generation phase, it only extracts bindkey or/and token and writes it to the right place.

keniji, 2024-05-19:
> For this integration you need to configure both - bindkey and token. xiaomi_account is a helper that works only on code generation phase, it only extracts bindkey or/and token and writes it to the right place.

Got that, thanks 👍👍👍

closed 2024-05-19 by keniji