dentra/esphome-components #58: OTA changes in 2024.6.0-devbreaks `otax`

nagyrobi, 2024-05-21:
There will be a breaking change to OTA in the next release; the change was made to accommodate additional OTA platforms (in other words, you can OTA update in more than one way now!)
https://github.com/esphome/esphome/pull/6459
docs have been updated accordingly: https://next.esphome.io/components/ota
safe mode is also changing slightly: https://github.com/esphome/esphome/pull/6759
Discord thread: https://discord.com/channels/429907082951524364/1240771036249849866

Having
```yaml
external_components:
  - source: github://dentra/esphome-components

otax:
  enabled:
    name: ${friendly_name} OTA
```
in the config results in:
```
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/root/esphome/venv/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/__main__.py", line 1070, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/__main__.py", line 1048, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/config.py", line 1093, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/config.py", line 947, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/config.py", line 937, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/esphome/esphome/config.py", line 861, in validate_config
    result.run_validation_steps()
  File "/root/esphome/esphome/config.py", line 144, in run_validation_steps
    task.step.run(self)
  File "/root/esphome/esphome/config.py", line 740, in run
    self.comp.final_validate_schema(conf)
  File "/root/config/.esphome/external_components/2f2a5be3/components/otax/__init__.py", line 37, in _final_validate
    ota_conf[CONF_ID].type = Otax
    ~~~~~~~~^^^^^^^^^
TypeError: list indices must be integers or slices, not str
```

dentra, 2024-05-22:
try with latest changes

dentra, 2024-05-22:
it still hacky and just overrides original implementation, but can be implemented as separate ota platform in future

dentra, 2024-05-24:
I've made some fix, you can check it out?

nagyrobi, 2024-05-24:
I'll try this week-end. Need to reflash it by serial first, as OTA is blocked now :.)

nagyrobi, 2024-05-24:
OK now on both ESP32 and 8266.

dentra, 2024-05-26:
👍

closed 2024-05-26 by dentra