dentra/esphome-components #59: Energy Tariffs Component

Roving-Ronin, 2024-08-03:
Hi @dentra,

Thanks for these components, using the Energy Statistic component with my Athom v2 and v3 plugs and it works quite well.
When using webserver version = 3, with its ability to allow defining the sort order of sensors, it comes out as:

[image]

Just having a play and looking to see if the Energy Tariffs component would be useful, but trying to work out if it can support the tariff plans we have in Australia. Leaving out the new 'Peak Demand' tariffs being introduced to the market, we have the following 'Time of Use' TOU plans that are typically:

Monday - Friday:
 Peak - 7am to 9am
 Shoulder - 9am to 5pm
 Peak - 5pm to 8pm
 Shoulder - 8pm to 10pm
 Off Peak - 10pm to 7am

Saturday and Sundays:
 Off Peak - All Day/Night (i.e. from 10pm Friday to 7am Monday)

Is your Tariff component able to support these tariffs, or does the component currently only support defining the time but not allow specifying the day also ?

Many Thanks !

Roving-Ronin, 2024-08-03:
Ooops... I should have checked closed issues first. Just noticed this old issue:  https://github.com/dentra/esphome-components/issues/39

```
      - name: "P1"
        time: "06:00-14:00"
        days_of_week: &workdays MON-FRI
        exclude: &holidays [25.12, 26.12, 27.12]
      - name: "P2"
        time: "14:00-22:00"
        days_of_week: *workdays
        exclude: *holidays
      - name: "P3"
```

Is there a version of the component that has the days_of_week functionality added to it that I can test? Looking at the C+ code I can't find any reference to this in the 'Master' branch version.

Thanks.

dentra, 2024-08-04:
#39 it is not implemented, but only proposed. If it fits, then I'll try to implement it.

Roving-Ronin, 2024-08-04:
Happy to test it, if you implement it ! ;-)
BTW where are the 'workdays' / 'holidays' / 'weekends' defined, or is that pulled from HomeAssistant and the WorkDay integration?

Roving-Ronin, 2024-08-14:
Looking at this component and its perfect to convert to provide a 'Water Statistics' also. Only issue is that in sensor.py it refers to kWh:

```
from esphome.const import (
    CONF_ID,
    CONF_TIME_ID,
    CONF_POWER,
    CONF_TOTAL,
    DEVICE_CLASS_ENERGY,
    STATE_CLASS_TOTAL_INCREASING,
    UNIT_KILOWATT_HOURS,
)
```
would you have any idea how to change the units to litres and it appears that esphome.const doesn't include this?

TIA.

dentra, 2024-08-15:
Is just setting unit_of_measurenent doesn't work?

Roving-Ronin, 2024-08-16:
Yes, I was having a go at it and that was where it failed. Have a look at: https://github.com/Roving-Ronin/myHomeAssistant/tree/main/components/water_statistics

If you want to copy that over to your repository and I'll delete my attempt, please do as it's your code and credit.

Looking at this, if cubic meters (m3) and megajoules (MJ) could also be supported, this would also allow you to publish a 'Gas Statistics (m3)' and 'Gas Statistics (MJ)' component also, so you'd have Energy, Gas and Water covered :-)
Anyone using MJ would need to be aware most utility companies update their 'Pressure Factor' and Heating Value' that is used to calculate m3 into MJ, and include a sensor for this in their ESPHome device that they update quarterly. i.e.

```

  - platform: template
    name: "Gas - Consumed (MJ)"
    id: gas_consumed_mj
    unit_of_measurement: "MJ"
    device_class: energy
    state_class: "total_increasing"
    icon: "mdi:counter"
    accuracy_decimals: 3   
    # With EACH QUARTERLY GAS BILL update the Jemena Pressure Factor and Heating Value below.   
    # Formula - (Gas Consumed x Pressure Factor x Heating Value) = Usage MJ
    lambda: |-
      return (id(gas_consumed_m3).state * 0.9864 * 37.98);
    update_interval: 1s
    filters:
      - delta : 0.001
```

BTW any luck looking at adding 'Energy - Year' to Statistics or adding the days of week to Tariffs ( https://github.com/dentra/esphome-components/issues/39 ) ? Thanks !

Roving-Ronin, 2024-08-21:
Update: Managed to get this working for water (L) and gas (m3), will have to see if I can get it working for gas (mj) in future, Not as simple as converting gas from cubic meters to MJ involves calculation of the variables 'Pressure Factor' and 'Heating Value' that are dependent upon the gas provider and change regularly (quarterly for my provider).

Formula:  (Gas Consumed m3 x Pressure Factor x Heating Value) = MJ consumed

These are in: https://github.com/Roving-Ronin/myHomeAssistant/tree/main/components/ if you want to add into your repo.

BTW any luck looking at adding the days of week to Tariffs ( https://github.com/dentra/esphome-components/issues/39 ) ? Thanks !

dentra, 2026-10-13:
Implemented in 37a7c6f, available in `master`. Your TOU plan can now be configured like this (Off-Peak is the default tariff, so it covers nights and weekends):

```yaml
sensor:
  - platform: energy_tariffs
    total: total
    tariffs:
      - name: Peak
        time: ["07:00-09:00", "17:00-20:00"]
        days_of_week: MON-FRI
      - name: Shoulder
        time: ["09:00-17:00", "20:00-22:00"]
        days_of_week: MON-FRI
      - name: Off-Peak
```

About your question on workdays/holidays: there are two options.
- A static list on the device: `holidays: ["01.01", "26.01", "2026-04-03", ...]`. On a holiday, tariffs are selected as on Sunday (configurable with `holiday_as`).
- The HA [Workday](https://www.home-assistant.io/integrations/workday/) integration through the `holiday` option. Set all 7 days as workdays in Workday, so the sensor is `off` only on public holidays:

```yaml
binary_sensor:
  - platform: homeassistant
    id: holiday
    entity_id: binary_sensor.workday_sensor
    filters:
      - invert:

sensor:
  - platform: energy_tariffs
    total: total
    holiday: holiday
    tariffs:
      ...
```

Details are in the [README](https://github.com/dentra/esphome-components/tree/master/components/energy_tariffs#days-of-week-and-holidays). If you find a problem or have a suggestion, please open a new issue.

closed 2026-10-13 by dentra