dentra/esphome-components #61: How to use crash_info? xtensa-lx106-elf-addr2line: file format not recognized

TN-1, 2024-08-12:
Hi all,

I am trying to debug a crash with some ESPHome devices I am using. I have captured the stack trace, however I am having trouble feeding it into the decoder.

I have tried copying the full trace as presented, isolating just the trace frames, and several differents formats of the afformentioned however I keep getting the error as in the title.

Could someone please clarify the format that is expected? My google-fu is failing me here.

Trace:
```
[09:18:06][I][crash_info:067]: Crash info: 
[09:18:06][I][crash_info:070]:   Exception cause code: 3, time: 2024-08-11 18:46:31
[09:18:06][I][crash_info:078]:   Stacktrace:
[09:18:06][I][crash_info:096]:      40245d57 4000050c 40242461 40000f68
[09:18:06][I][crash_info:096]:      4023fc8a 401010da 40242940 40242940
```

Thanks in advance!

TN-1, 2024-08-12:
I have made some progress, I wasnt inlduing the .elf file. Now, I am getting the following output:
0x00000000: ?? ??:0

closed 2024-10-19 by stale[bot]

dentra, 2026-10-13:
Sorry for the late reply. `0x00000000: ?? ??:0` means addr2line got a zero address, usually because the log lines were passed as is. Pass only the addresses as arguments and use the firmware.elf from the same build that crashed:

```
xtensa-lx106-elf-addr2line -aipfC -e .esphome/build/<node>/.pioenvs/<node>/firmware.elf \
  40245d57 4000050c 40242461 40000f68 4023fc8a 401010da 40242940 40242940
```

Addresses starting with 4000 are in ESP8266 ROM, `??` is expected for them.

Also note that crash_info is unsupported since ESPHome 2026.4.0: ESPHome now has a built-in ESP8266 crash handler, and `esphome logs` decodes the backtrace automatically.