dentra/esphome-components #65: Energy Statistics - Update Week/Month for Partial Time Period

Roving-Ronin, 2025-01-02:
Hi, @dentra 

Currently the energy_statistics sensor only starts counting for the week and month, if the device is running as it changes from Sunday to Monday / from the last day of the month to the first day of the month.  So for example if you plug in a power sensor on day 2 of the week/month (or even part way through the first day) it won't update the weekly or monthly sensors, instead you have to wait for another week/month to pass before they start update.

Would you know how to change this, so these sensors all initalize and start couning immediately, like the today sensor?

Thanks.

Roving-Ronin, 2025-01-06:
Worked it out, in .CPP change the Process code to:

```
void EnergyStatistics::process_(float total) {
  // Publish today's energy
  if (this->energy_today_ && !std::isnan(this->energy_.start_today)) {
    this->energy_today_->publish_state(total - this->energy_.start_today);
  } else if (std::isnan(this->energy_.start_today)) {
    // Initialize today's start value
    this->energy_.start_today = total;
    if (this->energy_today_) {
      this->energy_today_->publish_state(0);  // Publish initial value as 0
    }
  }

  // Publish yesterday's energy
  if (this->energy_yesterday_ && !std::isnan(this->energy_.start_yesterday)) {
    this->energy_yesterday_->publish_state(this->energy_.start_today - this->energy_.start_yesterday);
  }

  // Publish weekly energy (partial or full)
  if (this->energy_week_ && !std::isnan(this->energy_.start_week)) {
    if (this->energy_.full_week_started) {
      // Publish full calendar week value
      this->energy_week_->publish_state(total - this->energy_.start_week);
    } else {
      // Publish partial week value
      this->energy_week_->publish_state(total - this->energy_.start_week);
    }
  } else if (std::isnan(this->energy_.start_week)) {
    // Initialize start_week for the first time
    this->energy_.start_week = total;
    if (this->energy_week_) {
      this->energy_week_->publish_state(0);  // Publish initial value as 0
    }
  }

  // Publish monthly energy (partial or full)
  if (this->energy_month_ && !std::isnan(this->energy_.start_month)) {
    if (this->energy_.full_month_started) {
      // Publish full calendar month value
      this->energy_month_->publish_state(total - this->energy_.start_month);
    } else {
      // Publish partial month value
      this->energy_month_->publish_state(total - this->energy_.start_month);
    }
  } else if (std::isnan(this->energy_.start_month)) {
    // Initialize start_month for the first time
    this->energy_.start_month = total;
    if (this->energy_month_) {
      this->energy_month_->publish_state(0);  // Publish initial value as 0
    }
  }
 
  // Save the current state
  this->save_();
}
```

And in the .h change the definitions in energy_data_t to add the full week/month, whilst the existing are used for partial time periods:

```
  struct energy_data_t {
    uint16_t current_day_of_year{0};
    float start_today{NAN};
    float start_yesterday{NAN};
    float start_week{NAN};
    float start_month{NAN};
    bool full_week_started{false};    // Added
    bool full_month_started{false};  // Added
    bool full_year_started{false};   // Added
  } energy_;  // Instance of the struct
```

@dentra any interest in this? If so, will create a PULL request for you...?

dentra, 2025-01-14:
please have a look update at master branch

Roving-Ronin, 2025-01-15:
You seems to have done it a bit different, looking at energy_statistics.cpp though, are you missing:

```
    // at first day of year we start a new year calculation
    if (t.day_of_year == 1) {
      this->energy_.start_year = total;
    }
```
around line 87 - https://github.com/dentra/esphome-components/blob/91e438dbb6e4f07290bac7762c7bdcddea3a67f4/components/energy_statistics/energy_statistics.cpp#L87

Also in energy_statistics.h would you need:

```
  // start day of year configuration
  int energy_year_start_day_{1};
```

Roving-Ronin, 2025-01-15:
Just testing your v2 energy_statistics and using your current 'master' code, it appears NOT to be updating all the sensors. The only sensors that are updating are the Energy Today and Total (i.e. lifetime) sensors. 
Given its Wednesday, it would appear the Weekly / Monthly / Yearly sensors are not updating as they are still expecting to only start updating come 0:00am on Monday / 1st of Month / 1st of January.

Here's a screenshot of a new Athom Smart Plug (No Relay) that I just flashed with your v2 master code and ran a toaster on to get the sensors to update:

[image]

As you can see, the Week/Month/Year are not updating for a partial time period. If you want to, suggest having a look at the updated version of this code that I'm running at: https://github.com/Roving-Ronin/myHomeAssistant/tree/main/components/energy_statistics

This is successfully running with all sensors updating, here's an example that was reflashed on ~4th Jan 2025, to get it working with the month / year updating (and the week also, but its now rolled to having full weeks) even though its a partial month/year.

[image]

As a side issue, in the past I found that if/when updating the firmware due to a change or new release of ESPHome, all the sensors would lose their data. Given this, in the yaml I've found it best to use the sensors from the custom_components to be 'internal: true' and have a lambda that updates a global. From this I then have corresponding template sensors, that read the globals and publish the values. Using this approach means that the today / yesterday / week / month / year and total sensors are all stored by the globals and this data persists between reflashing/updating firmware.

If of interest, here is the templated yaml that I use for all my Athom Smart Plugs (no relay) models, there's also versions for Smart Plug v2 and v3 (that have relays), see:  https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/sensors/athom-power-plug-no-relay.yaml

PS. Probably possible to have this functionality to save to global then publish via sensor built into the custom_component, buy just did it in yaml as it was faster and easier.

Roving-Ronin, 2025-01-17:
Just testing your v2 energy_statistics and using your current 'master' code, it appears NOT to be updating all the sensors. 

Edit:

Took a while to update the week / month sensors, but they are now updating. However I note that the Year sensor is NOT updating. I would assume this as the code for year is missing in places within the .cpp and .h

[image]

Please refer to PULL 67 for patches: https://github.com/dentra/esphome-components/pull/67
______________________________________________________________________

As a side issue, in the past I found that if/when updating the firmware due to a change or new release of ESPHome, all the sensors would lose their data. Given this, in the yaml I've found it best to use the sensors from the custom_components to be 'internal: true' and have a lambda that updates a global. From this I then have corresponding template sensors, that read the globals and publish the values. Using this approach means that the today / yesterday / week / month / year and total sensors are all stored by the globals and this data persists between reflashing/updating firmware.

If of interest, here is the templated yaml that I use for all my Athom Smart Plugs (no relay) models, there's also versions for Smart Plug v2 and v3 (that have relays), see:  https://github.com/Roving-Ronin/myHomeAssistant/blob/main/esphome/sensors/athom-power-plug-no-relay.yaml

PS. Probably possible to have this functionality to save to global then publish via sensor built into the custom_component, buy just did it in yaml as it was faster and easier.

dentra, 2025-01-18:
I do not use year stats, so I might have missed something. Thanks. I will look your PR.

My code uses absolutely the same approach to store data as globals do. Are you using "restore_from_flash: true" for  your esp8266? Also "flash_write_interval" may matters when you reflash faster than it was set.

Roving-Ronin, 2025-01-19:
Yes on ESP8285 I have:
```
esp8266:
  board: esp8285
  restore_from_flash: true
```

Whilst ESP8285 and ESP32 both have:
```
preferences:
   # Enable to allow storing of 'Read Total' between reboots and flashing.
  flash_write_interval: 5min
```

Still erases at times with reflashing / updating ESPHome versions on them, whereas the Energy Total sensor (that is feed into the component) that is stored as a global remains safe, only wiping with a factory reset of the device.

Roving-Ronin, 2025-01-20:
@dentra  I just updated to the latest version / merge and see 'NA' for week, month & year. Could you please confirm that your seeing these all update constantly i.e. during partial week / month / year time periods?

Roving-Ronin, 2025-01-20:
As I'm only seeing my 2 test plugs both not update for week/month/year as flashing with your latest code. Just comparing code and with this part of .h extended with the bool for full_week/month/year_started, that then has the float versions being used for the partial time periods:

```
  struct energy_data_t {
    uint16_t current_day_of_year{0};
    float start_today{NAN};
    float start_yesterday{NAN};
    float start_week{NAN};
    float start_month{NAN};
    float start_year{NAN};
    bool full_week_started{false};    // Added
    bool full_month_started{false};  // Added
    bool full_year_started{false};   // Added
  } energy_;  // Instance of the struct
```

Wouldn't your .cpp need to have something like this for the process section, with it add 'if its a new week/month/year etc then start counting, else if its a partial week/month/year then count that, else publish 0' ?

```
void EnergyStatistics::process_(float total) {
  // Publish today's energy
  if (this->energy_today_ && !std::isnan(this->energy_.start_today)) {
    this->energy_today_->publish_state(total - this->energy_.start_today);
  } else if (std::isnan(this->energy_.start_today)) {
    // Initialize today's start value
    this->energy_.start_today = total;
    if (this->energy_today_) {
      this->energy_today_->publish_state(0);  // Publish initial value as 0
    }
  }

  // Publish yesterday's energy
  if (this->energy_yesterday_ && !std::isnan(this->energy_.start_yesterday)) {
    this->energy_yesterday_->publish_state(this->energy_.start_today - this->energy_.start_yesterday);
  }

  // Publish weekly energy (partial or full)
  if (this->energy_week_ && !std::isnan(this->energy_.start_week)) {
    if (this->energy_.full_week_started) {
      // Publish full calendar week value
      this->energy_week_->publish_state(total - this->energy_.start_week);
    } else {
      // Publish partial week value
      this->energy_week_->publish_state(total - this->energy_.start_week);
    }
  } else if (std::isnan(this->energy_.start_week)) {
    // Initialize start_week for the first time
    this->energy_.start_week = total;
    if (this->energy_week_) {
      this->energy_week_->publish_state(0);  // Publish initial value as 0
    }
  }

  // Publish monthly energy (partial or full)
  if (this->energy_month_ && !std::isnan(this->energy_.start_month)) {
    if (this->energy_.full_month_started) {
      // Publish full calendar month value
      this->energy_month_->publish_state(total - this->energy_.start_month);
    } else {
      // Publish partial month value
      this->energy_month_->publish_state(total - this->energy_.start_month);
    }
  } else if (std::isnan(this->energy_.start_month)) {
    // Initialize start_month for the first time
    this->energy_.start_month = total;
    if (this->energy_month_) {
      this->energy_month_->publish_state(0);  // Publish initial value as 0
    }
  }

  // Publish yearly energy (partial or full)
  if (this->energy_year_ && !std::isnan(this->energy_.start_year)) {
    if (this->energy_.full_year_started) {
      // Publish full calendar year value
      this->energy_year_->publish_state(total - this->energy_.start_year);
    } else {
      // Publish partial year value
      this->energy_year_->publish_state(total - this->energy_.start_year);
    }
  } else if (std::isnan(this->energy_.start_year)) {
    // Initialize start_year for the first time
    this->energy_.start_year = total;
    if (this->energy_year_) {
      this->energy_year_->publish_state(0);  // Publish initial value as 0
    }
  }
```

Roving-Ronin, 2025-01-22:
No idea why, but after 1.5 - 2 days the plugs have now started updating and showing the Week / Month kWh readings. 
Year readings are still showing 'NA', but have just worked out that this is because the .cpp file at the Process component is missing the task to tell it to update the Year sensor.

I've submitted a PULL to correct this omission: https://github.com/dentra/esphome-components/pull/70

Roving-Ronin, 2025-01-23:
@dentra, Would having this code in the 'EnergyStatistics::loop' instead of the 'EnergyStatistics::process_' section be the cause of why the week/month/year sensors are taking 36-48 hours to first update?

```
  // Intitialize all sensors. https://github.com/dentra/esphome-components/issues/65
  if (this->energy_week_ && std::isnan(this->energy_.start_week)) {
    this->energy_.start_week = this->energy_.start_yesterday;
  }
  if (this->energy_month_ && std::isnan(this->energy_.start_month)) {
    this->energy_.start_month = this->energy_.start_yesterday;
  }
  if (this->energy_year_ && std::isnan(this->energy_.start_year)) {
    this->energy_.start_year = this->energy_.start_yesterday;
  }
```

Roving-Ronin, 2025-01-23:
As it appears the Yesterday value initially being NAN (NA in web interface) does cause the week / month / year sensors to not update, till the yesterday sensor received a numerical value. Thinking we could simply default Yesterday value to 0 (IF there is no prior value in globals), by using this for the Process component:

```
void EnergyStatistics::process_(float total) {
  if (this->energy_today_ && !std::isnan(this->energy_.start_today)) {
    this->energy_today_->publish_state(total - this->energy_.start_today);
  }

  if (this->energy_yesterday_ && !std::isnan(this->energy_.start_yesterday)) {
    this->energy_yesterday_->publish_state(this->energy_.start_today - this->energy_.start_yesterday);
  } else if (this->energy_yesterday_) {
    // If there's no value for yesterday (NaN), publish 0
    this->energy_yesterday_->publish_state(0);
  }

  if (this->energy_week_ && !std::isnan(this->energy_.start_week)) {
    this->energy_week_->publish_state(total - this->energy_.start_week);
  } else if (this->energy_week_) {
    // If there's no value for week (NaN), publish 0
    this->energy_week_->publish_state(0);
  }

  if (this->energy_month_ && !std::isnan(this->energy_.start_month)) {
    this->energy_month_->publish_state(total - this->energy_.start_month);
  } else if (this->energy_month_) {
    // If there's no value for month (NaN), publish 0
    this->energy_month_->publish_state(0);
  }

  if (this->energy_year_ && !std::isnan(this->energy_.start_year)) {
    this->energy_year_->publish_state(total - this->energy_.start_year);
  } else if (this->energy_year_) {
    // If there's no value for year (NaN), publish 0
    this->energy_year_->publish_state(0);
  }

  this->pref_.save(&this->energy_);
}
```

Testing it on a few plugs and code PLUS the PULL https://github.com/dentra/esphome-components/pull/70 has it working all correctly.

dentra, 2025-01-26:
Yes, the year, month and week will update on next day, it is by design to optimize calculation time. NaN is showing that this info is unavailable yet.

Roving-Ronin, 2025-01-26:
This PULL still needs to be merged anyhow, as your missing code for the Year function

https://github.com/dentra/esphome-components/pull/70

closed 2025-04-03 by stale[bot]